Interbotix/interbotix_ros_rovers #8: [Enhancement]:

EdSeymore, 2021-12-14:
### Describe the solution you'd like

Provide a universal launch file at the Application layer which would support the navigation and perception stacks and allow exercising all components of the LoCoBot with python applications. 

### Platform

xslocobots

### Is your feature request related to a problem? Please describe.

This would remove the current complexity of having several launch files.

EdSeymore, 2021-12-14:
Thanks, I had overlooked this launch file's additional capabilities after using for the bartender sample. I suggest using this file in place of xslocobot_control.launch on https://www.trossenrobotics.com/Shared/RTD/XSeries_Locobots/docs/ROS_Interface/Software_Setup.html#installation-checks Quickstart. Also, perhaps adding joystick functions as well for a first motion test.

LSinterbotix, 2021-12-20:
I like the idea of adding the joystick control check to the quickstart. Thanks for the suggestion!

If you have no further questions on this issue, you can feel free to close it.

EdSeymore, 2021-12-22:
If a change is made to  interbotix_xslocobot_control/xslocobot_python.launch, how will users be notified?

LSinterbotix, 2021-12-22:
We don't have a formal notification process yet, but you can periodically fetch changes to see if any updates were made to the remote branches:
```bash
cd ~/interbotix_ws/src/interbotix_ros_rovers && git fetch
```

If changes were made, you can update your local repo by merging the changes:
```bash
cd ~/interbotix_ws/src/interbotix_ros_rovers && git merge
```

Or you can do this in one shot by pulling:
```bash
cd ~/interbotix_ws/src/interbotix_ros_rovers && git pull
```

Note that you should always back up or stash your changes before pulling from the remote repo.

EdSeymore, 2022-01-07:
I have not taken the time to analyze all of the defined launch files; But, my impression is that  interbotix_xslocobot_control/xslocobot_python.launch does not include all options to support a fully functional locobot with capabilities to run all sample applications at the same time. I do not see the advantage of having such a complex set of launch files to combine in order to get a fully functional LoCoBot.

EdSeymore, 2022-01-10:
I remember seeing a chart of all of the locobot launch files and how they are related at some time since I received mine. Search as I may, I have not been unable to locate it again. Could you provide a link to that chart? I did not realize at the time how important that chart was. Given that I am still trying to identify the simplest way to support all capabilities of locobot from a single launch file at the same time, I am still struggling with my configuration. (Problem seems to be the small area which I have for navigation functions.) 

Suggestion: copy xslocobot_python.launch into locobot_master.launch and use that in the initial setup and run examples since it seems to be a good way to support all locobot functions. I got sidetracked by starting with control launch. Also, the initial quickstart step 7 should not use geometry_msgs/Twist. w/o a warning that the robot may run into objects.  It would be safer to start with the PS4 controller so that all operations are under human control. At least change the yaw to 0 so it goes straight ahead. 

Again thanks for the extensive help and a really great product.

EdSeymore, 2022-01-12:
OK, I found it at https://www.trossenrobotics.com/docs/interbotix_xslocobots/ros_packages/python_demos.html. 

BTW: Your team is doing a great job of making the web site easier to navigate for the new user.

closed 2022-01-12 by EdSeymore

LSinterbotix, 2022-01-12:
Thanks for the suggestions!

We've updated the ROS Interface > Software Setup > [Quickstart](https://www.trossenrobotics.com/docs/interbotix_xslocobots/ros_interface/software_setup.html#quickstart) section with a warning about the robot moving like we did with torquing the arm off causing it to fall. We also mention using the joystick control package at bit further down in step 11.