Interbotix/interbotix_ros_rovers #9: [Question]:

EdSeymore, 2021-12-18:
### Question

What is the function of the locobot static transform in static_transforms.yaml? 
Value is: 
static_transforms.yaml: 
----------------------- 
cd /home/locobot/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_perception/config/ 
nano static_transforms.yaml 
------------------------------------------------------ 
- child_frame_id: locobot/plate_link
  frame_id: locobot/base_link
  qw: 1.0
  qx: -5.613034173113181e-16
  qy: 1.0552620243447992e-16
  qz: 5.727731113791439e-16
  x: -0.11701889308097978
  y: 0.035955274369297706
  z: 0.4097284782215129
------------------------------------------------------ 

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 20.04 
ROS noetic

### Anything Else

tune the perception pipeline fails due too Dist Thresh value just does not seem to be working.

LSinterbotix, 2021-12-20:
The static_transforms.yaml configuration file is created as a result of calibrating the arm's pose relative to the rest of the robot using the [ArmTag Tuner GUI](https://github.com/Interbotix/interbotix_ros_rovers/tree/main/interbotix_ros_xslocobots/interbotix_xslocobot_perception#armtag-calibration). The file stores the transform (translation and rotation) between the LoCoBot's base_link and the plate_link. Looking at your transform, it seems that something may have gone wrong when calibrating the ArmTag. Try running the ArmTag Calibration process again. The transform should be small, on the order of a few mm in each xyz direction.

Note that this process is separate from the PointCloud Filter tuning process that you mention in your issue's Anything Else section. I would suggest creating a new issue for that.

EdSeymore, 2021-12-22:
You are correct: The values cited were after using the wrong robot model. 
My most recent values after renaming static_transforms.yaml are: 
12/21/21 14:54 nano static_transforms.yaml 

- child_frame_id: locobot/plate_link
  frame_id: locobot/base_link
  qw: 1.0
  qx: 7.330976037501461e-17
  qy: 2.311323022943178e-16
  qz: -9.609172219209353e-17
  x: 0.001984563576085771
  y: 0.003189061402826529
  z: 0.09514801603860834 

My new questions are: 
What is the purpose of the ArmTag tuner GUI? 
Why is static_transforms.yaml z value so large [0.095]? 
Observation: When following process in https://www.youtube.com/watch?v=UesfMYM4qcc 
My objects and the floor are floating above the locobot/base_footprint level. 
Note: I using LoCoBot WX250 6 DOF with robot_model:=locobot_wx250s.

LSinterbotix, 2021-12-22:
- **What is the purpose of the ArmTag tuner GUI?**

The ArmTag Tuner GUI is the tool that creates the static_transforms.yaml configuration file and publishes that transform between base_link and plate_link. Because the transform between the camera and the arm is known (since everything is connected through rigid bodies), you don't actually need to use the tool unless you want very fine precision, as mentioned in the [xslocobot_perception usage](https://github.com/Interbotix/interbotix_ros_rovers/tree/main/interbotix_ros_xslocobots/interbotix_xslocobot_perception#usage) section:

> However, since the arm and camera in this case are part of the same robot, this transform is already known from the URDF. So there really is no need to use the AR tag on the arm to figure out this transform. The tag is mainly there so that if you decide to remove the arm from the Locobot for some other project, you can still use the perception pipeline.

- **Why is static_transforms.yaml z value so large [0.095]?**

I'm not sure why it would ever be that large - there may have been some error in the calibration process. You can either delete the file, or rerun the calibration process.

- **My objects and the floor are floating above the locobot/base_footprint level**

This may be due to small inaccuracies in the URDF building up to slightly increase the height of the camera, resulting in an increase in the height of your workpieces. This might also happen if your objects are slightly higher up in the workspace relative to the base footprint. The incorrect static transform config may also have something to do with it.

EdSeymore, 2021-12-28:
I am still unclear on why anyone would need to use ArmTag Calibration. None of the procedures seem to make any adjustments based upon the results of running that process. I also do not understand why base_link and plate_link were selected for the locobot case. The arm sits on the plate_link. But, the base_link does not seem to have any particular relationship to the arm tag or the camera positions. The fact that on my system, the objects are above the floor level is very troubling. Also, other than the crop box filter, none of the other controls seem to make any difference to the Rviz display.

LSinterbotix, 2021-12-29:
You're right - the calibration is unnecessary on the LoCoBots since the transform between the camera and the arm is already known through the URDF.

The base and plate links were chosen as the calibration target because it's not a particularly important transformation and only serves to refine the position of the arm's base relative to the rest of the robot.

The objects not sitting at floor level may not matter if the transformations between the arm, camera, and objects are all correct. You can test this by running one of the pick and place scripts, or doing measurements on your own.

As for the RViz displays, you should make sure that you're only displaying the ObjectCloud and ObjectMarkers displays under the Perception folder. The FilteredCloud display is only affected by the crop box and the voxel resolution filters. See the explanation [here](https://youtu.be/UesfMYM4qcc?t=1149).

EdSeymore, 2021-12-29:
Do you have a LoCoBot WX250 6 DOF on which the Perception Pipeline Tuning actually works? If so, please build a new Youtube video specifically for this device. My Rviz does not match the https://www.youtube.com/watch?v=UesfMYM4qcc video.  Please show the operation at actual speed, my configuration does not seems to be responsive. I am using Intel NUC NUC10i7FNH 6-Core i7 for the remote Ubuntu 20.04 image. 

I am unable to remove the floor from the filtered output. My device/configuration is not working.

EdSeymore, 2022-01-04:
While I still have more questions than answers at this point: I have gotten most of the sample applications to run. 

The biggest problem with perception tuning was failing to understand that the floor is not removed from the FilteredCloud Rviz display. Displaying only the ObjectMarkers was the key to moving forward. 

I also needed to change ref_frame from base_link to arm_base_link to get pick_place_no_armtag to actually move the objects.  

It would have saved me a great deal of time if there was a chart on https://github.com/Interbotix/interbotix_ros_manipulators/tree/main/interbotix_ros_xsarms/interbotix_xsarm_perception showing the values of ref_frame for xslocobot_perception.launch and armtag.launch for the various camera/arm configurations. 

Thanks for your time in answering my questions. Your team has a great product in the LoCoBot WX250 6 DOF.

closed 2022-01-04 by EdSeymore