Interbotix/interbotix_ros_rovers #10: [Question]: How to charge the LoCoBot base

EdSeymore, 2021-12-23:
### Question

What are the important things to know about charging the base battery? 
Given: charge when the LED is amber. 
Q. Should the base be turned to ON? [seems to be required]
Q. When is the battery fully charged? [one LED turns from amble to green quickly] 
Q. What does the green LED on the charger indicate? [seems to be on when every attached to power] 

### Robot Model

LoCoBot WX250 6 DOF [mobile_wx250s] 

### Operating System

n/a

### Anything Else

Would like for this information to be included on https://www.trossenrobotics.com/docs/interbotix_xslocobots/getting_started/index.html

LSinterbotix, 2021-12-27:
The current mobile base of the LoCoBot is Yujin Robot's iClebo Kobuki. You can find its user guide [here](http://kobuki.yujinrobot.com/wiki/online-user-guide/).

To answer your questions:

1. The base does not need to be turned on when it is charging.
2. The battery is fully charged when the Status LED is solid green.
3. You are correct - the charger's LED is on when it is connected to power.

We will add some of this information to the documentation.

closed 2021-12-27 by EdSeymore

LSinterbotix, 2022-01-12:
We added some of this information to the new Getting Started > [Basic Usage](https://www.trossenrobotics.com/docs/interbotix_xslocobots/getting_started/user_guide.html) page.