Interbotix/interbotix_ros_rovers #11: [Question]: Is it permissible to use the processor while the rectangular battery is charging

EdSeymore, 2021-12-23:
### Question

Is it permissible to use the processor for coding and examining the code while the rectangular battery is charging?

### Robot Model

LoCoBot WX250 6 DOF [locobot_wx250s]

### Operating System

n/a

### Anything Else

I would prefer that is information be included with the getting started material at https://www.trossenrobotics.com/docs/interbotix_xslocobots/getting_started/index.html.

LSinterbotix, 2021-12-27:
Yes, you are able to charge the external battery and use the NUC computer at the same time. We will include this information in the documentation.

closed 2021-12-27 by EdSeymore

LSinterbotix, 2022-01-12:
We added this information to the new Getting Started > [Basic Usage](https://www.trossenrobotics.com/docs/interbotix_xslocobots/getting_started/user_guide.html) page.