Interbotix/interbotix_ros_rovers #12: [Bug]:

ninja-hu, 2022-01-15:
### What happened?

Using gazebo simulation on the LoCoBot fails. The LoCoBot keeps falling after the gazebo physics is unpaused. Following the instructions on the video tutorial page here : https://www.trossenrobotics.com/docs/interbotix_xslocobots/python_interface/index.html   

### Robot Model

locobot_px100

### Operating System

Ubuntu 16.04

### Steps To Reproduce

Follow the steps in the video tutorial on this page: 
https://www.trossenrobotics.com/docs/interbotix_xslocobots/python_interface/index.html   

to start the gazebo simulation on the LoCoBot NUC. Once the physics is unpaused , robot keeps falling.

ninja-hu, 2022-01-19:
HI @LSinterbotix , I am replicating the gazebo simulation part. I used the same command as mentioned in the video at 2:43 to launch the gazebo simulation. After that when I unpause gazebo's physics engine, the locobot keeps falling (behaves as if there is no floor below it.) 

I tried the same on a different system which had Ubuntu 20.04 (Noetic) and everything worked perfectly. The problem could be specific to Kinetic.

LSinterbotix, 2022-01-19:
I'm not able to replicate this issue on ROS Kinetic.

A few questions for you:

- Are you on the `kinetic` branch for the interbotix_ros_rovers repo?
- Did you clone the repos manually, or did you use the install script?
- Does the robot fall all the way through the ground plane, or stop halfway through?
- Could you provide a video or gif of this occurring? Have you modified the URDF or mesh files at all?

LSinterbotix, 2022-01-28:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2022-01-28 by LSinterbotix