Interbotix/interbotix_ros_rovers #13: [Question]: Trouble with rpLidar Node in a Gazebo simulation

Federico-Maresca, 2022-04-08:
### Question

Hi! I am trying to use the interbotix gazebo package together with the interbotix_nav package to run SLAM however the rpLidar node keeps crashing and rviz can't seem to receive any sensor data. Is there a guide or tutorial that runs through how this would be possible? If not how come?

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 20.04

LSinterbotix, 2022-04-09:
Hi @Federico-Maresca. That would be because we apparently never put a lidar plugin in the [Gazebo configs](https://github.com/Interbotix/interbotix_ros_rovers/blob/2858f6053f7c2a5782821c427779feccdcc8a05d/interbotix_ros_xslocobots/interbotix_xslocobot_gazebo/config/locobot_configs.gazebo). I went ahead and added the lidar along with a few other changes. You should be able to pull the changes to your local interbotix_ros_rovers repo.

I also added the [xslocobot_nav_sim](https://github.com/Interbotix/interbotix_ros_rovers/blob/2858f6053f7c2a5782821c427779feccdcc8a05d/interbotix_ros_xslocobots/interbotix_xslocobot_nav/launch/xslocobot_nav_sim.launch) launch file that should instruct you how to properly launch the navigation package in Gazebo. To launch it with your setup, use the command below:

```console
roslaunch interbotix_xslocobot_nav xslocobot_nav_sim.launch robot_model:=locobot_wx250s dof:=6 use_lidar:=true
```

Let me know if you run into any difficulties.

Federico-Maresca, 2022-04-09:
Thank you for the quick reply, the issue has been solved !

closed 2022-04-09 by Federico-Maresca