Interbotix/interbotix_ros_rovers #15: [Bug]: Waist motor position is incorrect, is a factory reset in the dynamixel wizard advisable?

dazednconfusing, 2022-05-18:
### What happened?

The waist position is -0.78 radians about when it should be zero. On the remote_view launch node, the arm is displayed at -0.78 radians incorrectly. 

I do not believe this is a mechanical issue but I cannot see the waist motor as it is covered.

I tried rebooting all the motors with the rosservice "reboot_motors" but it did not help. I am unable to calibrate in dynamixal wizard 2.0 as I get a "calibration bootloader error" message. All firmware is up to date.

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 18.04

### Steps To Reproduce

Unclear

dazednconfusing, 2022-05-18:
I solved the issue. The dynamixel calibration tool works if you only have one motor connected and you unplug and replug power immediately (don't wait a few seconds like it asks). I however calibrated incorrectly (mixed up 12 o' clock and 6 o' clock) but after factory resetting it automatically fixed.

I believe this issue was indeed caused by the waist being rotated 360 degrees at some point but I cannot be sure.

LSinterbotix, 2022-05-18:
Glad to hear that you were able to solve it. Feel free to reopen if you have further problems with this.

closed 2022-05-18 by LSinterbotix