Interbotix/interbotix_ros_rovers #17: [Bug]: Can't launch Gazebo simulation

Dorteel, 2022-05-25:
### What happened?

Trying to follow the [tutorials](https://www.youtube.com/watch?v=xIril2gF0-Y&t=71s), but when loading gazebo with the command:
`roslaunch interbotix_xslocobot_gazebo xslocobot_gazebo.launch robot_model:=locobot_wx250s show_lidar:=true use_position_controllers:=true dof:=6` I get an error as the `controller_spawner` doesn't have a unique name.

Tried changing the name of the trajectory controller node to something else, but then the arm of the robot wouldn't load.

Do you have any suggestions?

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 20.04

### Steps To Reproduce

`roslaunch interbotix_xslocobot_gazebo xslocobot_gazebo.launch robot_model:=locobot_wx250s show_lidar:=true use_position_controllers:=true dof:=6`

### Relevant log output

```shell
(base) user@dorteel:~$ roslaunch interbotix_xslocobot_gazebo xslocobot_gazebo.launch robot_model:=locobot_wx250s show_lidar:=true use_position_controllers:=true dof:=6
... logging to /home/user/.ros/log/8d0bf2d6-dc05-11ec-8729-61df804d5638/roslaunch-dorteel-18852.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: roslaunch file contains multiple nodes named [/locobot/controller_spawner].
Please check all <node> 'name' attributes to make sure they are unique.
Also check that $(anon id) use different ids.
The traceback for the exception was written to the log file
```

### Anything Else

Here is the output when the controller name is changed:

```console
(base) user@dorteel:~$ roslaunch interbotix_xslocobot_gazebo xslocobot_gazebo.launch robot_model:=locobot_wx250s show_lidar:=true use_position_controllers:=true dof:=6
... logging to /home/user/.ros/log/8d0bf2d6-dc05-11ec-8729-61df804d5638/roslaunch-dorteel-19712.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.15.3.55:37123/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /locobot/arm_controller/constraints/elbow/goal: 0.2
 * /locobot/arm_controller/constraints/elbow/trajectory: 0.2
 * /locobot/arm_controller/constraints/forearm_roll/goal: 0.2
 * /locobot/arm_controller/constraints/forearm_roll/trajectory: 0.2
 * /locobot/arm_controller/constraints/goal_time: 0.2
 * /locobot/arm_controller/constraints/shoulder/goal: 0.2
 * /locobot/arm_controller/constraints/shoulder/trajectory: 0.2
 * /locobot/arm_controller/constraints/waist/goal: 0.2
 * /locobot/arm_controller/constraints/waist/trajectory: 0.2
 * /locobot/arm_controller/constraints/wrist_angle/goal: 0.2
 * /locobot/arm_controller/constraints/wrist_angle/trajectory: 0.2
 * /locobot/arm_controller/constraints/wrist_rotate/goal: 0.2
 * /locobot/arm_controller/constraints/wrist_rotate/trajectory: 0.2
 * /locobot/arm_controller/gains/elbow/d: 1.0
 * /locobot/arm_controller/gains/elbow/i: 10.0
 * /locobot/arm_controller/gains/elbow/p: 300.0
[... 119 lines cut ...]
[ INFO] [1653469260.665846467]: Loaded gazebo_ros_control.
[ INFO] [1653469260.670646436]: Starting plugin DiffDrive(ns = /locobot//)
[ INFO] [1653469260.670695254]: DiffDrive(ns = /locobot//): <rosDebugLevel> = na
[ INFO] [1653469260.671042432]: DiffDrive(ns = /locobot//): <tf_prefix> = 
[ INFO] [1653469260.671551472]: DiffDrive(ns = /locobot//): Advertise joint_states
[ INFO] [1653469260.671902646]: DiffDrive(ns = /locobot//): Try to subscribe to mobile_base/commands/velocity
[ INFO] [1653469260.672706140]: DiffDrive(ns = /locobot//): Subscribe to mobile_base/commands/velocity
[ INFO] [1653469260.672888886]: DiffDrive(ns = /locobot//): Advertise odom on odom 
[locobot/urdf_spawner-6] process has finished cleanly
log file: /home/user/.ros/log/8d0bf2d6-dc05-11ec-8729-61df804d5638/locobot-urdf_spawner-6*.log
[INFO] [1653469260.909780, 0.000000]: wait_for_service(/locobot/controller_manager/load_controller): finally were able to contact [rosrpc://10.15.3.55:56891]
[INFO] [1653469260.911020, 0.000000]: wait_for_service(/locobot/controller_manager/load_controller): finally were able to contact [rosrpc://10.15.3.55:56891]
Loaded 'arm_controller'
Loaded 'joint_state_controller'
Loaded 'gripper_controller'
Loaded 'waist_controller'
Loaded 'pan_controller'
Loaded 'shoulder_controller'
Loaded 'tilt_controller'
[ERROR] [1653469261.037935207]: A controller named 'joint_state_controller' was already loaded inside the controller manager
Loaded 'elbow_controller'
Error when loading 'joint_state_controller'
Loaded 'forearm_roll_controller'
```
[image]

LSinterbotix, 2022-05-25:
I can't reproduce this on a fresh source install of the xslocobot packages on Noetic using your roslaunch command:

<details>
  <summary>roslaunch console output</summary>

```sh
~ $ roslaunch interbotix_xslocobot_gazebo xslocobot_gazebo.launch robot_model:=locobot_wx250s show_lidar:=true use_position_controllers:=true dof:=6
... logging to /home/interbotix/.ros/log/f53d995a-dc4a-11ec-b898-47855316a363/roslaunch-interbotix-728613.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.183:40927/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /locobot/elbow_controller/joint: elbow
 * /locobot/elbow_controller/pid/d: 0.0
 * /locobot/elbow_controller/pid/i: 0.0
 * /locobot/elbow_controller/pid/p: 300.0
 * /locobot/elbow_controller/type: effort_controller...
 * /locobot/forearm_roll_controller/joint: forearm_roll
 * /locobot/forearm_roll_controller/pid/d: 0.0
 * /locobot/forearm_roll_controller/pid/i: 0.0
 * /locobot/forearm_roll_controller/pid/p: 50.0
 * /locobot/forearm_roll_controller/type: effort_controller...
 * /locobot/joint_state_controller/publish_rate: 100
 * /locobot/joint_state_controller/type: joint_state_contr...
 * /locobot/left_finger_controller/joint: left_finger
 * /locobot/left_finger_controller/pid/d: 5.0
 * /locobot/left_finger_controller/pid/i: 0.0
 * /locobot/left_finger_controller/pid/p: 50
 * /locobot/left_finger_controller/type: effort_controller...
 * /locobot/pan_controller/joint: pan
 * /locobot/pan_controller/pid/d: 0.1
 * /locobot/pan_controller/pid/i: 1.0
 * /locobot/pan_controller/pid/p: 30.0
 * /locobot/pan_controller/type: effort_controller...
 * /locobot/right_finger_controller/joint: right_finger
 * /locobot/right_finger_controller/pid/d: 5.0
 * /locobot/right_finger_controller/pid/i: 0.0
 * /locobot/right_finger_controller/pid/p: 50
 * /locobot/right_finger_controller/type: effort_controller...
 * /locobot/robot_description: <?xml version="1....
 * /locobot/shoulder_controller/joint: shoulder
 * /locobot/shoulder_controller/pid/d: 0.0
 * /locobot/shoulder_controller/pid/i: 0.0
 * /locobot/shoulder_controller/pid/p: 500
 * /locobot/shoulder_controller/type: effort_controller...
 * /locobot/tilt_controller/joint: tilt
 * /locobot/tilt_controller/pid/d: 0.1
 * /locobot/tilt_controller/pid/i: 1.0
 * /locobot/tilt_controller/pid/p: 30.0
 * /locobot/tilt_controller/type: effort_controller...
 * /locobot/waist_controller/joint: waist
 * /locobot/waist_controller/pid/d: 0.0
 * /locobot/waist_controller/pid/i: 0.0
 * /locobot/waist_controller/pid/p: 100
 * /locobot/waist_controller/type: effort_controller...
 * /locobot/wrist_angle_controller/joint: wrist_angle
 * /locobot/wrist_angle_controller/pid/d: 0.0
 * /locobot/wrist_angle_controller/pid/i: 0.0
 * /locobot/wrist_angle_controller/pid/p: 50.0
 * /locobot/wrist_angle_controller/type: effort_controller...
 * /locobot/wrist_rotate_controller/joint: wrist_rotate
 * /locobot/wrist_rotate_controller/pid/d: 0.0
 * /locobot/wrist_rotate_controller/pid/i: 0.0
[... 87 lines cut ...]
</details>

In your roslaunch console output, it seems like both the position and trajectory controllers are trying to load since you have both the arm/gripper and the individual joint controller parameters. I was able to get your error if I try to launch both the trajectory and position controllers at the same time (setting [both these args](https://github.com/Interbotix/interbotix_ros_rovers/blob/c71b0c969a3324cd261470c63abb2ca56984cc7a/interbotix_ros_xslocobots/interbotix_xslocobot_gazebo/launch/xslocobot_gazebo.launch#L18-L19) to true).

<details>
  <summary>RLException</summary>

```sh
~ $ roslaunch interbotix_xslocobot_gazebo xslocobot_gazebo.launch robot_model:=locobot_wx250s show_lidar:=true dof:=6 use_position_controllers:=true use_trajectory_controllers:=true
... logging to /home/interbotix/.ros/log/7a4c6cf8-dc4a-11ec-b898-47855316a363/roslaunch-interbotix-728173.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: roslaunch file contains multiple nodes named [/locobot/controller_spawner].
Please check all <node> 'name' attributes to make sure they are unique.
Also check that $(anon id) use different ids.
The traceback for the exception was written to the log file
```

</details>

This would try to load the controller_spawner node twice ([here](https://github.com/Interbotix/interbotix_ros_rovers/blob/c71b0c969a3324cd261470c63abb2ca56984cc7a/interbotix_ros_xslocobots/interbotix_xslocobot_gazebo/launch/xslocobot_gazebo.launch#L40-L47) and [here](https://github.com/Interbotix/interbotix_ros_rovers/blob/c71b0c969a3324cd261470c63abb2ca56984cc7a/interbotix_ros_xslocobots/interbotix_xslocobot_gazebo/launch/xslocobot_gazebo.launch#L76-L86)), leading to the RLException you're seeing.

Check to see if you're launching both sets of controllers.

LSinterbotix, 2022-06-08:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2022-06-08 by LSinterbotix