Interbotix/interbotix_ros_rovers #18: [Bug]: dynamixel ID 2, 11 not found

aaravrav142, 2022-07-21:
### What happened?

After installation of the software on the NUC on the locobot_wx250 and launching with the following command 
`roslaunch interbotix_xslocobot_control xslocobot_control.launch robot_model:=locobot_wx250s use_base:=true use_camera:=true use_lidar:=true`  

I am getting the following response. Basically the dynamixel ID 2, 11 are not found. Earlier I had the same issue with the 10 and 11 so I checked with the Dynamixel wizard tool and found that the pan tilt motors were on 57600 baud rate. As instructed I changed the baud rate to 1 MBPS. Soon after this, the ID 2 and 11 disappeared from the scan list.  Also the realsense is not getting recognized on my NUC (New robot system). 
`locobot@locobot:~$ roslaunch interbotix_xslocobot_control xslocobot_control.launch robot_model:=locobot_wx250s use_base:=true use_camera:=true use_lidar:=true
... logging to /home/locobot/.ros/log/017a18f2-08b9-11ed-ae27-cbc681687424/roslaunch-locobot-4134.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.242:40471/

SUMMARY
========

PARAMETERS
 * /locobot/camera/realsense2_camera/accel_fps: 250
 * /locobot/camera/realsense2_camera/accel_frame_id: locobot/camera_ac...
 * /locobot/camera/realsense2_camera/accel_optical_frame_id: locobot/camera_ac...
 * /locobot/camera/realsense2_camera/align_depth: False
 * /locobot/camera/realsense2_camera/aligned_depth_to_color_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_infra1_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_infra2_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/allow_no_texture_points: False
 * /locobot/camera/realsense2_camera/base_frame_id: locobot/camera_link
 * /locobot/camera/realsense2_camera/calib_odom_file: 
 * /locobot/camera/realsense2_camera/clip_distance: -2.0
 * /locobot/camera/realsense2_camera/color_fps: 30
 * /locobot/camera/realsense2_camera/color_frame_id: locobot/camera_co...
 * /locobot/camera/realsense2_camera/color_height: 480
 * /locobot/camera/realsense2_camera/color_optical_frame_id: locobot/camera_co...
 * /locobot/camera/realsense2_camera/color_width: 640
 * /locobot/camera/realsense2_camera/depth_fps: 30
 * /locobot/camera/realsense2_camera/depth_frame_id: locobot/camera_de...
 * /locobot/camera/realsense2_camera/depth_height: 480
 * /locobot/camera/realsense2_camera/depth_optical_frame_id: locobot/camera_de...
 * /locobot/camera/realsense2_camera/depth_width: 640
 * /locobot/camera/realsense2_camera/device_type: 
 * /locobot/camera/realsense2_camera/enable_accel: False
 * /locobot/camera/realsense2_camera/enable_color: True
 * /locobot/camera/realsense2_camera/enable_depth: True
 * /locobot/camera/realsense2_camera/enable_fisheye1: False
 * /locobot/camera/realsense2_camera/enable_fisheye2: False
 * /locobot/camera/realsense2_camera/enable_fisheye: False
 * /locobot/camera/realsense2_camera/enable_gyro: False
 * /locobot/camera/realsense2_camera/enable_infra1: False
 * /locobot/camera/realsense2_camera/enable_infra2: False
 * /locobot/camera/realsense2_camera/enable_infra: False
 * /locobot/camera/realsense2_camera/enable_pointcloud: False
 * /locobot/camera/realsense2_camera/enable_pose: False
 * /locobot/camera/realsense2_camera/enable_sync: False
 * /locobot/camera/realsense2_camera/filters: 
 * /locobot/camera/realsense2_camera/fisheye1_frame_id: locobot/camera_fi...
 * /locobot/camera/realsense2_camera/fisheye1_optical_frame_id: locobot/camera_fi...
 * /locobot/camera/realsense2_camera/fisheye2_frame_id: locobot/camera_fi...
 * /locobot/camera/realsense2_camera/fisheye2_optical_frame_id: locobot/camera_fi...
 * /locobot/camera/realsense2_camera/fisheye_fps: 30
 * /locobot/camera/realsense2_camera/fisheye_frame_id: locobot/camera_fi...
 * /locobot/camera/realsense2_camera/fisheye_height: 480
 * /locobot/camera/realsense2_camera/fisheye_optical_frame_id: locobot/camera_fi...
 * /locobot/camera/realsense2_camera/fisheye_width: 640
 * /locobot/camera/realsense2_camera/gyro_fps: 400
 * /locobot/camera/realsense2_camera/gyro_frame_id: locobot/camera_gy...
 * /locobot/camera/realsense2_camera/gyro_optical_frame_id: locobot/camera_gy...
 * /locobot/camera/realsense2_camera/imu_optical_frame_id: locobot/camera_im...
[... 159 lines cut ...]
[locobot/xs_sdk-4] process has finished cleanly
log file: /home/locobot/.ros/log/017a18f2-08b9-11ed-ae27-cbc681687424/locobot-xs_sdk-4*.log
[ WARN] [1658382627.876197577]: No RealSense devices were found!
[ WARN] [1658382633.880016476]: No RealSense devices were found!
[ WARN] [1658382639.893466137]: No RealSense devices were found!
[ WARN] [1658382645.905147990]: No RealSense devices were found!
[ WARN] [1658382651.916776891]: No RealSense devices were found!
[ WARN] [1658382657.927400359]: No RealSense devices were found!
[ WARN] [1658382663.941613925]: No RealSense devices were found!
[ WARN] [1658382669.952314678]: No RealSense devices were found!
[ WARN] [1658382675.963724925]: No RealSense devices were found!
[ WARN] [1658382681.974209903]: No RealSense devices were found!
`

### Robot Model

locobot_wx250 

### Operating System

Ubuntu 20.04

### ROS Distro

ROS1 Noetic

LSinterbotix, 2022-07-26:
You may have an ID conflict somewhere along the chain. Try to remove the arm's 3-pin cable from the power hub so that the U2D2 can only see the pan/tilt two-axis servo. Run the Dynamixel Wizard, scan using the typical search parameters, and see if you can find the missing servos. If you can, set the IDs and baudrates according to the [default servo configuration](https://www.trossenrobotics.com/docs/interbotix_xslocobots/specifications/locobot_wx250s.html#default-servo-configurations).

To the realsense issue, try running the command `realsense-viewer` command to see if librealsense can detect the camera without the ROS wrapper.

aaravrav142, 2022-08-01:
Closing this for now. The problem with the Realsense still exists.

closed 2022-08-01 by aaravrav142