Interbotix/interbotix_ros_rovers #20: [Question]: Offset source in the perception pipeline?

daniel-gheorghita, 2022-09-16:
### Question

I encountered two slightly weird situations when following the Python Perception tutorial on Youtube. 
1. The gripper tries to grip correctly in RViz (goal EE frame on the object marker), but the arm moves at an offset to the object in the real world -> this I assumed to be a calibration problem. Then point (2) happened. 
2. The goal EE frame in RViz is computed at an offset (see the attached picture) and the arm goes at the same offset in the real world. Any ideas why this is happening and how it could be fixed?
[image]

Thank you!

### Robot Model

locobot_wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Noetic

LSinterbotix, 2022-10-19:
A few things:

1. Does your robot have the Create 3 base or the Kobuki base? Based on the RViz screenshot, it looks like the robot thinks it has the Kobuki base. If it has the Create 3 base instead, the arm is mounted slightly higher, which may cause this behavior.
2. Did you run the armtag calibration before the running your application?
3. Could you provide the contents of your terminal where you see this behavior?

daniel-gheorghita, 2022-10-26:
Hi, thank you for your reply! 
After I zeroed the motors with the Dynamixel Wizard I could not replicate the issue and the grasps seems successful.

closed 2022-10-26 by daniel-gheorghita