Interbotix/interbotix_ros_rovers #21: [Bug]: cannot install the amd64 install script

lxsy-xcy, 2022-09-21:
### What happened?

I want to install this package on my locobot , so I fellow the tutorial and run the script, but it failed and told me that can't find "$BRIDGE_MSGS_ROS1_WS/install_isolated/setup.bash".

I read the script and find that if I choose the kobuki base, I will not have that file, because BRIDGE_MSGS_ROS1_WS only have create3 base version.

And I wonder that if I choose ros1 to install, can I install the realsense by this script? I don't find the script to install that in ros1.

(I am not an English speaker and I hope I express my issue correctly.)

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 18.04

### ROS Distro

ROS1 Melodic

### Steps To Reproduce

./xslocobot_remote_install.sh -b kobuki 
(with 18.04 ubuntu and ros 1 melodic)

LSinterbotix, 2022-09-21:
This should be fixed with the above commit. Feel free to reopen this issue if not.

closed 2022-09-21 by LSinterbotix

lxsy-xcy, 2022-09-22:
I have another question about the script. I wonder that if I choose ros1 to install, can I install the realsense by this script? I don't find the script to install that in ros1.

LSinterbotix, 2022-09-22:
It is installed via rosdep since interbotix_xslocobot_control depends on it via its package.xml file:

https://github.com/Interbotix/interbotix_ros_rovers/blob/8a8af5addbb7256f1051a0e1b4d0e3cc59d2c5f2/interbotix_ros_xslocobots/install/amd64/xslocobot_amd64_install.sh#L328

https://github.com/Interbotix/interbotix_ros_rovers/blob/ea7bfb523ded49599e6eb4e4fafb9c4a07f77717/interbotix_ros_xslocobots/interbotix_xslocobot_control/package.xml#L19

lxsy-xcy, 2022-09-23:
thank you very much! I’ll try that.

lxsy-xcy, 2022-09-23:
Thank you very much! I’ll try that.

commit 8a8af5ad (2022-09-21): [ros_xslocobots] Fix install bug

Fix bug where all installs had to source the Create 3 messages workspace
which would not exist for Kobuki base installs.

Fixes #21
commit a45a0196 (2023-01-07): [ros_xslocobots] Fix install bug

Fix bug where all installs had to source the Create 3 messages workspace
which would not exist for Kobuki base installs.

Fixes #21
commit df88e30b (2023-01-07): [ros_xslocobots] Fix install bug

Fix bug where all installs had to source the Create 3 messages workspace
which would not exist for Kobuki base installs.

Fixes #21