Interbotix/interbotix_ros_rovers #22: [Question]: perception pipeline with an edited higher camera tower will get wrong point cloud,how can i fix that?

lxsy-xcy, 2022-10-01:
### Question

For some reason, we want to make the camera tower higher, and we did that.
[image]

But when we want to practice the perception pipeline, the point cloud just lower than expected.
[image]

And we wonder how to solve that problem.

(I am not an English speaker and I hope I express my issue clearly.)

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 18.04

### ROS Version

ROS1 Melodic

LSinterbotix, 2022-10-19:
Have you modified the URDF to include the higher supports and camera position? To do this, you should either add extra links to the URDF, or change the origin of the pan-and-tilt mechanism to reflect the changes you have made to the robot.

If you did modify the URDF, please send the contents of your console and the results of using the arm tag calibration GUI.

LSinterbotix, 2022-11-02:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2022-11-02 by LSinterbotix