Interbotix/interbotix_ros_rovers #24: [Bug]: spawner_configuration in xslocobot_nav_sim.launch causes "unused args" error

daniel-gheorghita, 2022-10-26:
### What happened?

Latest commit (06b6a9e9f3776e10bcba5039e19487bc4a103861) in xslocobot_gazebo.launch makes launching the xslocobot_nav_sim.launch cause an unused args error (because of "spawner_configuration"). 

Removing "spawner_configuration" from xslocobot_nav_sim.launch would fix the issue. 

### Robot Model

locobot_wx200

### Operating System

Ubuntu 20.04

### ROS Distro

ROS1 Noetic

### Steps To Reproduce

roslaunch xslocobot_nav_sim.launch

LSinterbotix, 2022-10-26:
Should be fixed in 5900ef01d693e6fbb4f59fb290555052a72afa6e. Let me know if this works for you.

daniel-gheorghita, 2022-11-02:
Yes, it works. Thank you!

closed 2022-11-02 by daniel-gheorghita