Interbotix/interbotix_ros_rovers #25: [Question]: Unable to completely reproduce the quickstart guide for wx250s

pkrishna037, 2022-12-07:
### Question

Hi,

I've been trying to reproduce the RViz based control of the wx250s arm without the base. To do this I ran the commands in the [ROS quickstart guide](https://docs.trossenrobotics.com/interbotix_xslocobots_docs/ros_interface/quickstart.html). But, I get the RViz window where the robot is white and with errors on `Global Status` and `RobotModel`. Would you be able to help on this?

I have changed the ROS_IP to be export ROS_IP=192.168.1.250 because the auto detect script set the ROS_IP as local_host.
[image]

## HOW TO REPRODUCE

I finished the setup on bot the locobot and the remote(which runs 20.04 with noetic) according to the [standard setup](https://docs.trossenrobotics.com/interbotix_xslocobots_docs/ros_interface/software_setup.html)

Then on remote I launch the `remote_view` following the `roslaunch interbotix_xslocobot_control xslocobot_control.launch`

I have given both commands below.

```
$ roslaunch interbotix_xslocobot_descriptions remote_view.launch --wait
```

### Remotely logged into NUC's gnome-terminal
```

locobot@locobot:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_descriptions/urdf/arms$ roslaunch interbotix_xslocobot_control xslocobot_control.launch robot_model:=locobot_wx250s use_base:=false use_camera:=true use_lidar:=false
... logging to /home/locobot/.ros/log/cfff7708-75e2-11ed-a0dc-7f1b06b615ec/roslaunch-locobot-17975.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.250:41323/

SUMMARY
========

PARAMETERS
 * /locobot/base_type: create3
 * /locobot/camera/realsense2_camera/accel_fps: -1
 * /locobot/camera/realsense2_camera/accel_frame_id: locobot/camera_ac...
 * /locobot/camera/realsense2_camera/accel_optical_frame_id: locobot/camera_ac...
 * /locobot/camera/realsense2_camera/align_depth: False
 * /locobot/camera/realsense2_camera/aligned_depth_to_color_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_infra1_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_infra2_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/allow_no_texture_points: False
 * /locobot/camera/realsense2_camera/base_frame_id: locobot/camera_link
 * /locobot/camera/realsense2_camera/calib_odom_file: 
 * /locobot/camera/realsense2_camera/clip_distance: -2.0
 * /locobot/camera/realsense2_camera/color_fps: 30
 * /locobot/camera/realsense2_camera/color_frame_id: locobot/camera_co...
 * /locobot/camera/realsense2_camera/color_height: 480
 * /locobot/camera/realsense2_camera/color_optical_frame_id: locobot/camera_co...
 * /locobot/camera/realsense2_camera/color_width: 640
 * /locobot/camera/realsense2_camera/confidence_fps: -1
 * /locobot/camera/realsense2_camera/confidence_height: -1
 * /locobot/camera/realsense2_camera/confidence_width: -1
 * /locobot/camera/realsense2_camera/depth_fps: 30
 * /locobot/camera/realsense2_camera/depth_frame_id: locobot/camera_de...
 * /locobot/camera/realsense2_camera/depth_height: 480
 * /locobot/camera/realsense2_camera/depth_optical_frame_id: locobot/camera_de...
 * /locobot/camera/realsense2_camera/depth_width: 640
 * /locobot/camera/realsense2_camera/device_type: 
 * /locobot/camera/realsense2_camera/enable_accel: False
 * /locobot/camera/realsense2_camera/enable_color: True
 * /locobot/camera/realsense2_camera/enable_confidence: True
 * /locobot/camera/realsense2_camera/enable_depth: True
 * /locobot/camera/realsense2_camera/enable_fisheye1: False
[... 146 lines cut ...]
 06/12 19:54:17,195 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:17,373 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:17,424 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:17,474 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:17,624 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:17,775 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:17,825 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:17,976 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:18,026 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:30,783 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/12 19:54:30,833 WARNING [140599504135936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Noetic

LSinterbotix, 2022-12-07:
Looks like RViz's fixed frame is set to `locobot/odom` which is unknown since the base is not being used. Try setting it to something like `locobot/base_link` instead.

```
roslaunch interbotix_xslocobot_descriptions remote_view.launch rviz_frame:=locobot/base_link
```

pkrishna037, 2022-12-07:
Thanks. That fixes it!

LSinterbotix, 2022-12-07:
Great! Feel free to close this issue if this issue is resolved.

closed 2022-12-07 by pkrishna037

psykaunot, 2024-06-11:
Hello, I have the same error but this solution does not work. I also saw one fixed issue where following the "IP" thing in the troubleshooting guide helped. However, none of them work for me. Even ` robot_state_publisher (robot_state_publisher/robot_state_publisher)` is not launched. 

I have to write it manually to make launch the node but still nothing. I even noticed that sometimes just by using `roslaunch interbotix_xslocobot_descriptions remote_view.launch rviz_frame:=locobot/base_link` only the base of the robot is displayed in Rviz.

I don't know what happened but even the white envelope of the robot (as shown in the picture of this issue) is not displayed anymore.

lukeschmitt-tr, 2024-06-11:
@psykaunot Please open a new issue and provide screenshots, logs, etc.

psykaunot, 2024-06-12:
@lukeschmitt-tr  I am waiting for you to see the issue #69  before closing this