Interbotix/interbotix_ros_rovers #27: [Question]: Gripper won't open to the max limit

aaravrav142, 2022-12-23:
### Question

Hello, 
I am using the Locobot wx250s model on ROS noetic. When using the joystick control for operating the arm using the command
`roslaunch interbotix_xslocobot_joy xslocobot_joy.launch robot_model:=locobot_wx250s controller:=ps4`
The gripper does not open completely. the closing part is not a problem. I though it could be because of the slider getting stuck between the rails and added some lubricant but the problem persists. 

Can you suggest how can I fix this problem or other ways  I should check if the gripper is operating fine. 

Thanks 

### Robot Model

locobot_wx250s 

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-01-03:
Could you provide more information about the issue including the following:

1. A picture/video of the behavior you're experiencing
2. What you expect the behavior to be
3. What you have tried to solve the issue (if anything other than adding the lubricant)
4. A `rostopic echo` of the LoCoBot's joint states at its "max"
5. If you have changed anything like the default configurations, launch files, etc.

LSinterbotix, 2023-01-09:
Closing due to lack of activity.

closed 2023-01-09 by LSinterbotix