Interbotix/interbotix_ros_rovers #28: Installation script is missing installing `interbotix_ros_core`

AVSurfer123, 2023-01-10:
The installation script that is used in the online documentation which is located [here](https://github.com/Interbotix/interbotix_ros_rovers/blob/main/interbotix_ros_xslocobots/install/xslocobot_remote_install.sh) is missing installing the [interbotix_ros_core](https://github.com/Interbotix/interbotix_ros_core) repo. This causes any other commands to fail, and it isn't explicitly stated that you need to install the core repo first for the rover repo to work. Either the documentation or the installation script should be updated to fix this.

LSinterbotix, 2023-01-10:
The rovers remote install is not meant to be built. Its only purpose is to provide the xslocobot_remote.launch file and the assets required to build the robot description (URDFs, xacros, mesh files, etc.), along with installing the distro-specific versions of the rtabmap_ros and xacro packages. That is why the `catkin_make` command is called before cloning the repos.

https://github.com/Interbotix/interbotix_ros_rovers/blob/863c2d60541938c0c189268d9fb0de354068d337/interbotix_ros_xslocobots/install/xslocobot_remote_install.sh#L231-L239

AVSurfer123, 2023-01-12:
Hmm I see. So the remote script assumes you only are using the `descriptions` and `gazebo` packages within this repo, which is why you don't build anything or need the dependencies. Thanks for clearing it up

closed 2023-01-12 by AVSurfer123