Interbotix/interbotix_ros_rovers #30: Gripper won't open to max (reopen)

aaravrav142, 2023-01-27:
Hi, 
Sorry for the late reply. I was travelling during the new year. Please check the video below to understand the problem. Basically, the gripper closes properly but needs some assistance when opening.  I do not have access to the bot right now. Once I return to the office, I can send the values of the joint state. Other than launching the sample joystick launch on the locobot, I did not change or do anything. Thanks,
https://photos.google.com/photo/AF1QipNL7wR2mzRXK8cr3NU7CZQUfnlEkY_AF2305e03

_Originally posted by @aaravrav142 in https://github.com/Interbotix/interbotix_ros_rovers/issues/27#issuecomment-1379840550_

LSinterbotix, 2023-02-03:
Hello @aaravrav142. That link is dead. Could you post an image directly to this issue instead?

aaravrav142, 2023-02-05:
Hi, 
Please try with the link below. I cannot directly share it here due to the large file size of the video. 
https://www.dropbox.com/s/5sro4jll52lwstn/trossen-gripper-issue.mp4?dl=0
Thanks

LSinterbotix, 2023-02-09:
Could you provide a video showing the face of the gripper motor and its swing arms?

swiz23, 2023-02-10:
Assuming you are using the interbitix_xslocobot_joy package, you could try raising the gripper pressure by hitting the Triangle button on your PS4 controller a few times (4 taps should raise it to the max from the default strength, or 8 taps if it's at its lowest strength). If that doesn't work, you could always spray a bit of WD-40 on a paper towel and coat the gripper rail with it.

aaravrav142, 2023-02-13:
Hi, 
Please give me a couple of days to report back. I am traveling for a conference. I will revert to the suggested tests as soon as possible. Thanks.

aaravrav142, 2023-02-17:
@swiz23 Your solution worked. Hitting the triangle button increased the pressure and the gripper moves smoothly now. Thanks a lot. I did not find this information in the documents though. It would be great if you can put this info as well in the locobot documentation and tutorial page. Many thanks for your help.

closed 2023-02-17 by aaravrav142