Interbotix/interbotix_ros_rovers #31: [Bug]: Missing tf link

stephenadhi, 2023-02-09:
### What happened?

Missing tf transformation from locobot/camera_tower_link to locobot/pan_link (frame somehow not detected). This causes the tf2 tree to split into two (see picture). 

[image]

### Robot Model

INTERBOTIX_XSLOCOBOT_BASE_TYPE=kobuki

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

### Steps To Reproduce

`ros2 launch interbotix_xslocobot_sim xslocobot_gz_classic.launch.py robot_model:=locobot_base robot_name:=locobot`

### Additional Info

Tried to directly load pan_and_tilt in locobot.urdf.xacro by removing the IF condition for camera_tower_size. Same tf2 tree result.

LSinterbotix, 2023-02-09:
Let me know if f619f996f8c0b5413d2c7c1af8728976a88ea23c and 8a9c29117dbfac5ba936a229f042d03c14a6cab4 fix this issue.

stephenadhi, 2023-02-09:
Yes, the missing link is solved now. Thanks for the quick response!

closed 2023-02-09 by stephenadhi