Interbotix/interbotix_ros_rovers #36: [Question]: Running Gazebo sim of LocoBot on remote computer

bkesari1998, 2023-03-01:
### Question

How can I install the correct Interbotix packages to run a Gazebo simulation of the LocoBot with the wx200 arm on a remote computer without having the ROS Master be a physical LocoBot.

I have tried running the remote install script (./xslocobot_remote_install.sh) and all this does is create the interbotix_ws with an empty src directory. 

I have also tried running the AMD install script and get this error:

/opt/ros/noetic/lib/python3/dist-packages/_distutils_hack/__init__.py:18: UserWarning: Distutils was imported before Setuptools, but importing Setuptools also replaces the `distutils` module in `sys.modules`. This may lead to undesirable behaviors or errors. To avoid these issues, avoid using distutils directly, ensure that setuptools is installed in the traditional way (e.g. not an editable install), and/or make sure that setuptools is always imported before distutils.
  warnings.warn(
/opt/ros/noetic/lib/python3/dist-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'interbotix_common_modules'
CMake Error at /opt/ros/noetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  
  execute_process(/home/bharatkesari/interbotix_ws/build/catkin_generated/env_cached.sh
  "/usr/bin/python3"
  "/opt/ros/noetic/share/catkin/cmake/interrogate_setup_dot_py.py"
  "interbotix_common_modules"
  "/home/bharatkesari/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_common_modules/setup.py"
  "/home/bharatkesari/interbotix_ws/build/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_common_modules/catkin_generated/setup_py_interrogation.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkin_python_setup.cmake:46 (safe_execute_process)
  interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_common_modules/CMakeLists.txt:17 (catkin_python_setup)

-- Configuring incomplete, errors occurred!
See also "/home/bharatkesari/interbotix_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/bharatkesari/interbotix_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
[ERROR] Failed to build ROS 1 packages for the Interbotix LoCoBot.
[ERROR] Interbotix Installation Failed!

### Robot Model

locobot_wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

### Additional Info

The physical LocoBot I have has the create3 base.

LSinterbotix, 2023-03-01:
See if the changes in 0e78282478eca9c03c0630a922e453e5fd6fa60e fix the empty src directory issue. Make sure to delete the interbotix_ws directory before running the remote install script again.

For the ROS master issue, you can just comment out the ROS_MASTER_URI line in your `~/.bashrc` file, or change it like below:

```diff
- export ROS_MASTER_URI=http://locobot.local:11311
+ export ROS_MASTER_URI=http://localhost:11311
```

bkesari1998, 2023-03-01:
Works now, thanks!

closed 2023-03-01 by bkesari1998