Interbotix/interbotix_ros_rovers #37: [Bug]: Nav stack issues when running in Gazebo

bkesari1998, 2023-03-03:
### What happened?

The navigation stack does not seem to be working in Gazebo for the locobot wx200.

This is the command I ran:

`roslaunch interbotix_xslocobot_nav xslocobot_nav.launch robot_model:=locobot_wx200 use_lidar:=true localization:=true`

[image]
[image]
[image]

### Robot Model

locobot_wx200

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 1 Noetic

### Steps To Reproduce

Running: `roslaunch interbotix_xslocobot_nav xslocobot_nav.launch robot_model:=locobot_wx200 use_lidar:=true localization:=true`

### Relevant log output

```shell
[ WARN] [1677871733.234596792, 268.547000000]: Could not get transform from locobot/base_footprint to locobot/camera_depth_link after 0.200000 seconds (for stamp=268.300000)! Error="Could not find a connection between 'locobot/base_footprint' and 'locobot/camera_depth_link' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201 timeout was 0.2.".
[ERROR] [1677871733.234641610, 268.547000000]: TF of received image 0 at time 268.300000s is not set!
[ERROR] [1677871733.234660496, 268.547000000]: Could not convert rgb/depth msgs! Aborting rtabmap update...
```

LSinterbotix, 2023-03-07:
It looks like no controllers were launched based on your Gazebo screenshot. Set either `use_position_controllers` or `use_trajectory_controllers` to `true` when launching the Gazebo simulation.

Note that we provide a launch file, [xslocobot_nav_sim.launch](https://github.com/Interbotix/interbotix_ros_rovers/blob/main/interbotix_ros_xslocobots/interbotix_xslocobot_nav/launch/xslocobot_nav_sim.launch), to assist with simulated navigation. This file will launch a Gazebo simulation and everything needed for the navigation stack.

bkesari1998, 2023-03-10:
Thanks, its working now!