Interbotix/interbotix_ros_rovers #41: [Bug]: Fatal Error during AMD64 installation on remote PC

Andrechief98, 2023-04-06:
### What happened?

Hello
I have previously performed the remote installation. After some trials, I decided for convenience to perform maximum network installation by following this video tutorial: 
[https://www.youtube.com/watch?v=zD4yg0cwiq4&t=1538s](url)

So, I removed the previous interbotix workspace and cleaned the related code in the bash.rc file (as in the video).

Following the command in the documentation, installation starts and create the ROS1 and ROS2 workspaces correctly. However, when it starts to build the ROS1 bridge, the installation completely crashes returning fatal error.

### Robot Model

locobot wx250 (create3 base)

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 1 Noetic

### Steps To Reproduce

``First I remove the "interbotix_ws" as in the video:

`$ sudo rm -r interbotx_ws/`

Then i remove all the code from the bash.rc file

Finally, I copy and paste the commands from the locobot documentation:

```
$ sudo apt install curl
$ curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_rovers/main/interbotix_ros_xslocobots/install/amd64/xslocobot_amd64_install.sh' > xslocobot_amd64_install.sh
$ chmod +x xslocobot_amd64_install.sh
$ ./xslocobot_amd64_install.sh -d noetic -b create3
```

This is the error:
[image]

Using this command in a new terminal:
`$ dmesg`
I obtain this:
[image]

### Additional Info

I try to search some possible related errors on Internet and It seems either a compiler bug or some memory problem. However, I have 8GB of memory on the remote pc. The last one seems strange to me.
I also try to modify a little bit the amd64 installation file adding "--parallel-workers 1" in the related colcon build command (line 470) in order to reduce the number of threads and don't overloading the memory but it dind't work.
I have no idea how to solve the problem.

thanks in advance for the help!

LSinterbotix, 2023-04-10:
I would agree that it looks like a memory issue as ros1_bridge is a very resource-intensive package to build. Looking around for solutions, I found [a ROS Answers thread](https://answers.ros.org/question/368106/c-fatal-moveit-2-error-killed-signal-terminated-program-cc1plus-compilation/?answer=373357#post-id-373357) stating that you may be able to provide some additional cmake arguments to colcon build, specifically,

```bash
colcon build --parallel-workers 1 --cmake-args -DCMAKE_CXX_FLAGS="--param ggc-min-expand=20"
```

You may also try setting the environment variable `MAKEFLAGS=-j1` before compilation as stated in the [ros1_bridge README](https://github.com/ros2/ros1_bridge#building-the-bridge-from-source).

Andrechief98, 2023-04-10:
Thanks for the answer. 
I modified the ams64.sh file and it finally finished the installation!

I copy and paste the code that I wrote (with the related line numbers) for help someone else that could have the same problem:
```
480 export MAKEFLAGS=-j1
481 echo ". /opt/ros/noetic/setup.bash && . /opt/ros/galactic/setup.bash && . $BRIDGE_MSGS_ROS1_WS/install_isolated/setup.bash && . $BRIDGE_MSGS_ROS2_WS/install/setup.bash && cd $BRIDGE_WS && colcon build --packages-select ros1_bridge --cmake-force-configure --parallel-workers 1 --cmake-args" > "$TEMP3"
```