Interbotix/interbotix_ros_rovers #44: [Question]: Unable to reproduce Landmark-Based Navigation

fyyffy, 2023-06-14:
### Question

I am trying to run the open source package [Landmark-Based Navigation](https://docs.trossenrobotics.com/interbotix_xslocobots_docs/ros1_packages/landmark_based_navigation.html).
Adding Landmarks and mapping are completed as per the instructions, but the commands under Localization are not executed as per the instructions.[image]

A new terminal opens but closes after outputting the following.

[image]
[image]

The log seems to recognize kobuki, but it is actually create3.
Please help if you know the cause.

### Robot Model

locobot_px100

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-06-15:
1. What commits are your interbotix_ros_core, ..._ros_rovers, and ..._ros_toolboxes repos at?
2. Check that the `/<robot_name/base_type` parameter is set to `create3` and not `kobuki`.
3. Check your `INTERBOTIX_XSLOCOBOT_BASE_TYPE` environment variable.

fyyffy, 2023-06-16:
1.Is this the correct information to provide?
[image]

[image]

[image]

2.The base_type parameter does not seem to exist. Is there a solution?
[image]

3.create3 is displayed.
[image]

Please let me know if you can find out anything.

LSinterbotix, 2023-06-26:
At the moment I would advise you to hardcore the base_type variable to `create3` in the [locobot.py](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/45cc26d6e2c82cc1cd2b649eee0ff7676f63b3d0/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/locobot.py#L60) file located in the traceback while we investigate this issue.

fyyffy, 2023-06-27:
locobot.py was an earlier version, so I fixed it and it worked without error. Thank you.

This video is of locobot in action, is the gradual movement normal? Is the sloping floor the cause?

https://github.com/Interbotix/interbotix_ros_rovers/assets/130592348/f5f7ae8d-6383-4494-a91d-6ed410aa5315

closed 2023-08-08 by fyyffy