Interbotix/interbotix_ros_rovers #45: [Question]: ROS 2 simulation + rtabmap

stephenadhi, 2023-06-20:
### Question

Hi, I'm wondering whether you plan to develop the locobot simulation and rtabmap launch in ROS 2?

`ros2 launch interbotix_xslocobot_sim xslocobot_gz_classic.launch.py robot_model:=locobot_base robot_name:=locobot use_lidar:=true use_rviz:=true use_gazebo_gui:=true `
This command didn't work as of now, missing camera topics.
`ros2 launch interbotix_xslocobot_nav xslocobot_rtabmap.launch.py  launch_driver:=false use_sim_time:=true`
[rgbd_sync-2] rgbd_sync subscribed to (exact sync):
[rgbd_sync-2]    /locobot/camera/color/image_raw,
[rgbd_sync-2]    /locobot/camera/aligned_depth_to_color/image_raw,
[rgbd_sync-2]    /locobot/camera/color/camera_info
[rtabmap-5] [WARN] [1687289649.932074396] [locobot.rtabmap.rtabmap]: rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10). 
[rtabmap-5] rtabmap subscribed to:
[rtabmap-5]    /locobot/rtabmap/rgbd_image

### Robot Model

locobot_base

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

stephenadhi, 2023-06-24:
Also currently in the simulation, the local costmap is drifting when the robot is navigating given a goal. This is not the case in real life. This is reproducible when you follow this [PR](https://github.com/Interbotix/interbotix_ros_rovers/pull/35)  using slam_toolbox (Step-to-step instructions updated)

[image]
[image]

aabs7, 2025-03-28:
Did you solve the drifting problem? I'm having the same drifting problem for locobot. Are the inertial values not setup properly?