Interbotix/interbotix_ros_rovers #46: [Bug]: Locobot wx200 with create3 base is drifting in Gazebo

goelshivam1210, 2023-07-05:
### What happened?

The locobot's navigation stack works well on the real robot. However, in simulation, we have two issues:
- When the robot moves forward, it drifts to the left. 
- When the robot is at rest, it's slowly rotating. 
We have checked the `cmd_vel` rostopic and confirmed that the velocity published is all zero when it's at rest. Thus, we suspect that maybe there is some issue in the gazebo file or the urdf? We're wondering if there's anything we can tune to solve this problem. Thanks!

### Robot Model

locobot_wx200 with create3 base

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 1 Noetic

### Steps To Reproduce

## Drifting to the Left:
We used the most bare-bone `xslocobot_gazebo.launch` file and can confirm the issue still exists. We ran the launch file by using this command:
```
roslaunch interbotix_xslocobot_gazebo xslocobot_gazebo.launch robot_model:=locobot_wx250s show_lidar:=true use_trajectory_controllers:=true
```
Then we started the simulation, and sent velocity to the `/locobot/cmd_vel` rostopic using [this teleop script](https://github.com/tufts-ai-robotics-group/locobot/blob/master/scripts/locobot_teleop.py).

## Rotating when at rest:
We used the official move_it launch file[2] that combined the move group and the navigation stack. We ran the following command:
```
roslaunch interbotix_xslocobot_moveit_interface xslocobot_moveit_interface.launch robot_model:=locobot_wx200 show_lidar:=true use_gazebo:=true use_cpp_interface:=true
```
We opened rviz and retracted the arm into the sleep position. Then we wait maybe 30 seconds, and the simulated robot started drifting by itself.

stephenadhi, 2023-07-10:
Hi,

I also have this exact same problem in ROS2 using the kobuki base, even though it's working seamlessly in real life. Robot drifting at rest without any command velocity.
[image]

goelshivam1210, 2023-07-10:
@stephenadhi Can you try the create3 base in ROS2? I could not get to fix the drifting of create3 base in ROS. However when the arm is in the home position, it stops drifting.

stephenadhi, 2023-07-13:
The ROS2 for create3 doesn't work properly. I suspect this idle drift problem comes from the inertia values. https://answers.gazebosim.org//question/25794/idle-robot-drifts-due-to-tilted-inertia/

aabs7, 2025-03-28:
Did you solve this problem? I'm having the same problem in ros2