Interbotix/interbotix_ros_rovers #48: [Bug]: Navigation Stack tutorial Bug

shanchaicn, 2023-07-26:
### What happened?

I tried to do the navigation with this command line:
```
roslaunch interbotix_xslocobot_nav xslocobot_nav.launch robot_model:=locobot_base use_lidar:=true rtabmap_args:=-d

```
I got some warning like this:
```
[ WARN] [1690398061.622297031]: 
 26/07 15:01:01,623 WARNING [140268917479168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:01:01,673 WARNING [140268917479168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:01:01,723 WARNING [140268917479168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:01:01,924 WARNING [140268917479168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:01:01,975 WARNING [140268917479168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:01:02,025 WARNING [140268917479168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1690398062.250737219]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690398061.976060)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201772 timeout was 0.2.".
[ WARN] [1690398062.393388031]: Timed out waiting for transform from locobot/base_footprint to map to become available before running costmap, tf error: Could not find a connection between 'map' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.100781 timeout was 0.1.
[ WARN] [1690398062.454532962]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690398062.048217)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201449 timeout was 0.2.".
[ WARN] [1690398062.658756609]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690398062.144190)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201429 timeout was 0.2.".
[ WARN] [1690398062.861524758]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690398062.217091)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201189 timeout was 0.2.".
[ WARN] [1690398063.074991094]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690398062.384497)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.202085 timeout was 0.2.".
[ WARN] [1690398063.278248217]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690398062.480565)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201888 timeout was 0.2.".

```

### Robot Model

locobot_base

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 1 Noetic

### Steps To Reproduce

After syncing the time with this issue[Navigation stack tutorial transform connection error](https://github.com/Interbotix/interbotix_ros_rovers/issues/43), it still didn't work.

### Relevant log output

```shell
locobot@locobot1:~$ roslaunch interbotix_xslocobot_nav xslocobot_nav.launch robot_model:=locobot_base use_lidar:=true rtabmap_args:=-d
... logging to /home/locobot/.ros/log/2f4ea0fa-2be9-11ee-b28c-b94d5dd7b39f/roslaunch-locobot1-9050.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.8.121:45975/

SUMMARY
========

PARAMETERS
 * /locobot/base_type: create3
 * /locobot/camera/realsense2_camera/accel_fps: -1
 * /locobot/camera/realsense2_camera/accel_frame_id: locobot/camera_ac...
 * /locobot/camera/realsense2_camera/accel_optical_frame_id: locobot/camera_ac...
 * /locobot/camera/realsense2_camera/align_depth: True
 * /locobot/camera/realsense2_camera/aligned_depth_to_color_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_infra1_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/aligned_depth_to_infra2_frame_id: locobot/camera_al...
 * /locobot/camera/realsense2_camera/allow_no_texture_points: False
 * /locobot/camera/realsense2_camera/base_frame_id: locobot/camera_link
[... 423 lines cut ...]
[ INFO] [1690399106.416577773]: SELECTED BASE:Depth, 0
[ INFO] [1690399106.426686836]: RealSense Node Is Up!
[ WARN] [1690399106.482301239]: 
 26/07 15:18:26,482 WARNING [140044102784768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:18:26,533 WARNING [140044102784768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:18:26,583 WARNING [140044102784768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:18:26,734 WARNING [140044102784768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:18:26,790 WARNING [140044102784768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:18:26,840 WARNING [140044102784768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 15:18:26,890 WARNING [140044102784768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1690399107.082522762]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690399106.809140)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.200337 timeout was 0.2.".
[locobot/camera_tilt-15] process has finished cleanly
log file: /home/locobot/.ros/log/2f4ea0fa-2be9-11ee-b28c-b94d5dd7b39f/locobot-camera_tilt-15*.log
[ WARN] [1690399107.286142511]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690399106.881471)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201221 timeout was 0.2.".
[ WARN] [1690399107.488504889]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690399106.977071)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201609 timeout was 0.2.".
[... cut ...]  part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201867 timeout was 0.2.".
[ WARN] [1690399109.109287496]: Could not get transform from locobot/odom to locobot/base_footprint after 0.200000 seconds (for stamp=1690399108.225845)! Error="Could not find a connection between 'locobot/odom' and 'locobot/base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201309 timeout was 0.2.".
```

LSinterbotix, 2023-07-26:
You may have unconnected trees because the robot is not able to see the Create 3's topics. I do not see any info about the tf_rebroadcaster publishing TFs so this may indeed be the case. 

Two things:
- Check that your [base's namespace](https://docs.trossenrobotics.com/interbotix_xslocobots_docs/getting_started/create3_configuration.html#main-configuration) is `/mobile_base`
- See the [Create® 3 base not responding to commands... troubleshooting section](https://docs.trossenrobotics.com/interbotix_xslocobots_docs/troubleshooting.html#create-3-base-not-responding-to-commands-can-t-see-topics-or-webserver-from-locobot-computer-ros-1-or-ros-2) for tips on solving this issue

shanchaicn, 2023-07-26:
Thank you very much! I forgot to change the Create3 config file. It works well now!

LSinterbotix, 2023-07-26:
Good to hear. Feel free to close this issue if your problem has been solved.

closed 2023-07-26 by shanchaicn