Interbotix/interbotix_ros_rovers #53: Create3 base movement and ps4 joypad

ajinkyamar, 2023-09-26:
### Question

Hi, 

I am trying to use a Sony PS4 wireless controller to move the base. However, I am unable to connect it via Bluetooth. 

The controller gets connected for less than a second and instantly, it is disconnected. 
I followed the instructions given in the documentation. I even tried installing some drivers, as suggested in some forums. 

Also, is there any other way to move the base without using a joypad? I couldn't find a command which is particular for locobot 250s. 

Can anyone please help me to resolve this? 

### Robot Model

locobot_wx250s with camera and rplidar

### Operating System

Ubuntu 20.04

### ROS Version

ROS 2 Galactic

ajinkyamar, 2023-09-27:
I checked again pairing controller documentation but it still persists, The PS4 controller gets connected for less than a second and it gets disconnected. I tried forgetting/removing the device but still no resolution.

How do I make use of the twist keyboard package?  I executed the above command in one terminal and tried pressing the given buttons after executing the joystick command (below) in a different terminal. 
ros2 launch interbotix_xslocobot_joy xslocobot_joy.launch.py robot_model:=locobot_wx250s

ajinkyamar, 2023-09-28:
How do I check that the cmd_vel topic the base subscribes to is /locobot/mobile_base/cmd_vel? 

The base is connected to locbot hotspot. I verified it using arp-a command. 
locobot@locobot:~$ arp -a
? (10.42.0.122) at <incomplete> on wlp0s20f3
? (10.42.0.60) at 74:d8:3e:a0:9b:aa [ether] on wlp0s20f3
? (192.168.186.2) at 4c:e1:73:42:1d:78 [ether] on eno1
locobot@locobot:~$ 

One device is my laptop, one is Bluetooth mouse. And the third one base, if my understanding is right.

LSinterbotix, 2023-09-28:
> `? (192.168.186.2) at 4c:e1:73:42:1d:78 [ether] on eno1`

The base is connected to the NUC via its Ethernet interface `eno1`.

You can the topic by running `ros2 topic list` while no xslocobot packages are running, looking for the topic with `cmd_vel` in it, and running `ros2 topic info -v <topic>` to see the subscribers. You should see the base's motion_control node subscribing to it.

```
$ ros2 topic list
...
/locobot/mobile_base/cmd_vel
...
$ ros2 topic info -v /locobot/mobile_base/cmd_vel
```

Additionally, you can use `rqt_graph` to see the connections between topics and nodes. This can be used to verify the topic subscriptions as well.

ajinkyamar, 2023-09-29:
Thanks! the base moved using keystroke. 

There is still problem with ps4 joypad Bluetooth connection. Same problem persists.

ajinkyamar, 2023-10-04:
https://github.com/Interbotix/interbotix_ros_rovers/issues/53#issuecomment-1737706972

How do I check joystick package to ensure that that aspect of the controller is not faulty, as mentioned above?  

I checked again the battery of the controller, charged it, and tried again.

LSinterbotix, 2023-10-04:
You can plug the controller directly into the LoCoBot's NUC when running the joystick control package. To the package, directly wiring the controller to the robot is the same as connection over Bluetooth.

ajinkyamar, 2023-10-09:
I tried wiring the controller as well. It is still not resolved. 

It shows 'ready to receive processed commands' and no command is accepted. 

locobot@locobot:~$ ros2 launch interbotix_xslocobot_joy xslocobot_joy.launch.py robot_model:=locobot_wx250s 
[INFO] [launch]: All log files can be found below /home/locobot/.ros/log/2023-10-09-08-57-22-681807-locobot-3540
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [3561]
[INFO] [joint_state_publisher-2]: process started with pid [3563]
[INFO] [rviz2-3]: process started with pid [3565]
[INFO] [xs_sdk-4]: process started with pid [3567]
[INFO] [component_container-5]: process started with pid [3569]
[INFO] [joy_node-6]: process started with pid [3571]
[INFO] [xslocobot_joy-7]: process started with pid [3573]
[INFO] [xslocobot_robot.py-8]: process started with pid [3575]
[xs_sdk-4] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.3'.
[xs_sdk-4] [INFO] Using logging level 'INFO'.
[xs_sdk-4] [INFO] Loaded mode configs from '/home/locobot/interbotix_ws/install/interbotix_xslocobot_joy/share/interbotix_xslocobot_joy/config/modes_all.yaml'.
[xs_sdk-4] [INFO] Loaded motor configs from '/home/locobot/interbotix_ws/install/interbotix_xslocobot_control/share/interbotix_xslocobot_control/config/locobot_wx250s.yaml'.
[xs_sdk-4] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID: 11, Model: 'XL430-W250-2', Joint Name: 'tilt'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  8, Model: 'XL430-W250', Joint Name: 'wrist_rotate'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  7, Model: 'XM430-W350', Joint Name: 'wrist_angle'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  5, Model: 'XM430-W350', Joint Name: 'elbow_shadow'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XM430-W350', Joint Name: 'elbow'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  6, Model: 'XM430-W350', Joint Name: 'forearm_roll'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XM430-W350', Joint Name: 'shoulder'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID: 10, Model: 'XL430-W250-2', Joint Name: 'pan'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  9, Model: 'XL430-W250', Joint Name: 'gripper'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XM430-W350', Joint Name: 'shoulder_shadow'.
[xs_sdk-4] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XM430-W350', Joint Name: 'waist'.
[component_container-5] [INFO] [1696859844.962335995] [tf_rebroadcaster_container]: Load Library: /home/locobot/interbotix_ws/install/interbotix_tf_tools/lib/libtf_rebroadcaster.so
[component_container-5] [INFO] [1696859844.964865036] [tf_rebroadcaster_container]: Found class: rclcpp_components::NodeFactoryTemplate<interbotix_tf_tools::TFRebroadcaster>
[component_container-5] [INFO] [1696859844.964900592] [tf_rebroadcaster_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<interbotix_tf_tools::TFRebroadcaster>
[component_container-5] [INFO] [1696859844.969074088] [tf_rebroadcaster]: Will broadcast TF from frame 'odom' to frame 'base_link', prepending prefix 'locobot/'.
[component_container-5] [INFO] [1696859844.969149698] [tf_rebroadcaster]: Will broadcast TF from frame 'odom' to frame 'base_footprint', prepending prefix 'locobot/'.
[component_container-5] [INFO] [1696859844.971528025] [tf_rebroadcaster]: Will broadcast TFs from topic '/locobot/mobile_base/tf' to the 'tf' topic under namespace '/'.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/tf_rebroadcaster' in container '/tf_rebroadcaster_container'
[xs_sdk-4] [WARN] Writing startup register values to EEPROM. This only needs to be done once on a robot if using a default motor config file, or after a motor config file has been modified. Can set `write_eeprom_on_startup` to false from now on.
[xs_sdk-4] [INFO] The operating mode for the 'arm' group was changed to 'position' with profile type 'time'.
[xs_sdk-4] [INFO] The operating mode for the 'camera' group was changed to 'position' with profile type 'velocity'.
[xs_sdk-4] [INFO] The operating mode for the 'gripper' joint was changed to 'pwm' with profile type 'velocity'.
[xs_sdk-4] [INFO] Interbotix X-Series Driver is up!
[xs_sdk-4] [INFO] [1696859846.182043781] [interbotix_xs_sdk.xs_sdk]: InterbotixRobotXS is up!
[xslocobot_robot.py-8] [INFO] [1696859846.362589623] [locobot.robot_manipulation]: 
[xslocobot_robot.py-8] 	Robot Name: locobot
[xslocobot_robot.py-8] 	Robot Model: locobot_wx250s
[xslocobot_robot.py-8] [INFO] [1696859846.362985462] [locobot.robot_manipulation]: Initialized InterbotixRobotXSCore!
[xs_sdk-4] [INFO] The operating mode for the 'pan' joint was changed to 'position' with profile type 'time'.
[xslocobot_robot.py-8] [INFO] [1696859846.471582907] [locobot.robot_manipulation]: Turret Group Name: camera
[xslocobot_robot.py-8] Pan Name: pan, Profile Type: time, Profile Velocity: 2.0, Profile Acceleration: 0.3
[xslocobot_robot.py-8] Tilt Name: tilt, Profile Type: time, Profile Velocity: 2.0, Profile Acceleration: 0.3
[xslocobot_robot.py-8] [INFO] [1696859846.471943273] [locobot.robot_manipulation]: Initialized InterbotixTurretXSInterface!
[xs_sdk-4] [INFO] The operating mode for the 'tilt' joint was changed to 'position' with profile type 'time'.
[xslocobot_robot.py-8] [INFO] [1696859847.086623753] [locobot.robot_manipulation]: Initialized InterbotixMobileBaseInterface!
[... cut ...]
[xslocobot_robot.py-8] [INFO] [1696859847.598498232] [locobot.robot_manipulation]: Initialized InterbotixArmXSInterface!
[xslocobot_robot.py-8] [INFO] [1696859848.107940094] [locobot.robot_manipulation]: 
[xslocobot_robot.py-8] 	Gripper Name: gripper
[xslocobot_robot.py-8] 	Gripper Pressure: 50.0%
[xslocobot_robot.py-8] [INFO] [1696859848.108315293] [locobot.robot_manipulation]: Initialized InterbotixGripperXSInterface!
[xslocobot_robot.py-8] [INFO] [1696859848.611692122] [locobot.robot_manipulation]: Ready to receive processed joystick commands.

LSinterbotix, 2023-10-11:
Does your controller light up when wired to the robot's computer?

Check for messages on the `/locobot/commands/joy_raw` and `/locobot/commands/joy_processed` topics when your controller is wired to the robot, you have launched the joystick control package, and you are giving inputs to the controller.

```
ros2 topic echo /locobot/commands/joy_raw
ros2 topic echo /locobot/commands/joy_processed
```

ajinkyamar, 2023-10-25:
yes, the controller lights up. I checked it on the windows system, it is connected via Bluetooth without any problem.

ajinkyamar, 2023-10-25:
**Below is another try with controller connected to NUC with wire.** 

This is new development. After "Ready to receive processed joystick commands", there are new processed commands/ broadcast messages (?) produced in the prompt. 

xslocobot_robot.py-8] [INFO] [1698236579.996287582] [locobot.robot_manipulation]: 
[xslocobot_robot.py-8] 	Arm Group Name: arm
[xslocobot_robot.py-8] 	Moving Time: 2.00 seconds
[xslocobot_robot.py-8] 	Acceleration Time: 0.30 seconds
[xslocobot_robot.py-8] 	Drive Mode: Time-Based-Profile
[xslocobot_robot.py-8] [INFO] [1698236579.996660630] [locobot.robot_manipulation]: Initialized InterbotixArmXSInterface!
[xslocobot_robot.py-8] [INFO] [1698236580.505946034] [locobot.robot_manipulation]: 
[xslocobot_robot.py-8] 	Gripper Name: gripper
[xslocobot_robot.py-8] 	Gripper Pressure: 50.0%
[xslocobot_robot.py-8] [INFO] [1698236580.506296629] [locobot.robot_manipulation]: Initialized InterbotixGripperXSInterface!
[xslocobot_robot.py-8] [INFO] [1698236581.009041710] [locobot.robot_manipulation]: Ready to receive processed joystick commands.
[component_container-5] [INFO] [1698236599.572030620] [tf_rebroadcaster]: Broadcasted TF from frame 'locobot/base_link' to frame 'locobot/odom'. This will only log once.
[component_container-5] [INFO] [1698236599.572200602] [tf_rebroadcaster]: Broadcasted TF from frame 'locobot/base_footprint' to frame 'locobot/odom'. This will only log once.

**Second thing:**
When I tried to echo joy_raw with below commands, the cursor just blinks without echoing/printing anything. 

ros2 topic echo /locobot/commands/joy_raw
ros2 topic echo /locobot/commands/joy_processed

What might be the issue here? 

Thanks