Interbotix/interbotix_ros_rovers #54: Rplidar not working

ajinkyamar, 2023-09-26:
### Question

Hi
I have issue with rplidar. 

When I do installation check with the below command, no output is produced. (rplidar as expected)

ls /dev | grep rplidar

I checked the wires, and power supply but still the problem persists. 

Can anyone please help me with this? 

### Robot Model

locobot_250s 

### Operating System

Ubuntu 20.04

### ROS Version

ROS 2 Galactic

LSinterbotix, 2023-09-26:
Check that the cables are plugged into the right ports:
- The power cable should be plugged into one of the USB ports on the battery.
- The data (short micro USB) cable should be plugged directly into the NUC.

ajinkyamar, 2023-09-27:
Thank you so much; it is working. 

I was using a longer cable to connect with NUC instead of the shorter micro USB.

closed 2023-09-27 by ajinkyamar