Interbotix/interbotix_ros_rovers #57: [Bug]: Tf has two or more unconnected trees.

michelalbonico, 2023-09-27:
### What happened?

Hi all,
I have been trying to run SLAM mapping on a simulated `wx250s` mobile robot.

However, it keeps reporting unconnected trees for `odom` and `base_link`, and therefore, no map is provided.

It seems that linking the `odom` to `base_link` in the robot's model should solve the problem. However, I could not manage it.

### Robot Model

simulated wx250s robot

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

### Steps To Reproduce

I have installed the necessary Interbotics packages for `humble` distro from source:
```
$ git clone -b humble-devel git@github.com:ROBOTIS-GIT/DynamixelSDK.git src/DynamixelSDK
$ git clone -b humble-devel git@github.com:ROBOTIS-GIT/dynamixel-workbench.git src/dynamixel-workbench
$ git clone -b humble git@github.com:Interbotix/interbotix_ros_core.git src/interbotix_ros_core
$ git clone -b ros2 git@github.com:Interbotix/interbotix_xs_driver.git src/interbotix_xs_driver
$ git clone -b humble git@github.com:Interbotix/interbotix_ros_manipulators.git src/interbotix_ros_manipulators
$ git clone -b humble git@github.com:IntelAgir-Research-Group/interbotix_ros_rovers.git src/interbotix_ros_rovers
$ git clone -b humble git@github.com:Interbotix/interbotix_ros_toolboxes.git src/interbotix_ros_toolboxes
$ git clone git@github.com:Slamtec/sllidar_ros2.git src/sllidar_ros2
```

Other dependency packages have been installed via `apt-get`.

I run the simulated robot with the following command:
```
$ ros2 launch interbotix_xslocobot_sim xslocobot_gz_classic.launch.py robot_model:=locobot_wx250s use_lidar:=true use_camera:=true
```

The simulated world and `rviz` run smoothly.

However, when I try slam mapping, it does not work properly (does not provide any map):
```
$ ros2 launch interbotix_xslocobot_nav xslocobot_rtabmap.launch.py robot_model:=locobot_wx250s use_lidar:=true slam_mode:=mapping rtabmap_args:=-d use_sim:=true
```

### Relevant log output

```shell
[controller_server-12] [INFO] [1695833553.672870393] [local_costmap.local_costmap]: Timed out waiting for transform from locobot/base_link to locobot/odom to become available, tf error: Could not find a connection between 'locobot/odom' and 'locobot/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[controller_server-12] [INFO] [1695833554.172893223] [local_costmap.local_costmap]: Timed out waiting for transform from locobot/base_link to locobot/odom to become available, tf error: Could not find a connection between 'locobot/odom' and 'locobot/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[controller_server-12] [INFO] [1695833554.672924179] [local_costmap.local_costmap]: Timed out waiting for transform from locobot/base_link to locobot/odom to become available, tf error: Could not find a connection between 'locobot/odom' and 'locobot/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
```

### Additional Info

Here is the graph containing the frame trees.
[frames.pdf.pdf](https://github.com/Interbotix/interbotix_ros_core/files/12741280/frames.pdf.pdf)

LSinterbotix, 2023-09-27:
Simulated navigation is not currently supported, though it is something we plan to add in the future.

michelalbonico, 2023-10-02:
Hi, 
I understand. The point is that we are bying a few Locobots for our research, and it is important that we have a simulated environment, such as for Turtlebots.

I will be plaing with this in the next weeks, and probably will come out with some solution.

michelalbonico, 2023-10-04:
This was a camera simulation issue. The `rtabmap` package uses the camera.

It worked with SLAM_toolbox after the following commands:
```
$ ros2 launch interbotix_xslocobot_sim xslocobot_gz_classic.launch.py robot_model:=locobot_wx250s use_rviz:=true use_lidar:=true use_camera:=false
$ ros2 launch interbotix_xslocobot_nav xslocobot_slam_toolbox.launch.py robot_model:=locobot_wx250s use_lidar:=true use_camera:=false slam_mode:=online_async use_sim:=true
```

closed 2023-10-04 by michelalbonico