Interbotix/interbotix_ros_rovers #68: Could not find a package configuration file provided by "interbotix_xs_sdk"

psykaunot, 2024-06-11:
### Question

Hello, 

I did `catkin_make` within the workspace because I am planning on adding other packages in your repository, but the following error appeared and I did not find a solution somewhere

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

### Additional Info

`locobot-admin@locobotadmin:~/interbotix_ws$ catkin_make
Base path: /home/locobot-admin/interbotix_ws
Source space: /home/locobot-admin/interbotix_ws/src
Build space: /home/locobot-admin/interbotix_ws/build
Devel space: /home/locobot-admin/interbotix_ws/devel
Install space: /home/locobot-admin/interbotix_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/locobot-admin/interbotix_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/locobot-admin/interbotix_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/locobot-admin/interbotix_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/locobot-admin/interbotix_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/locobot-admin/interbotix_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 17 packages in topological order:
-- ~~  - interbotix_common_modules
-- ~~  - interbotix_common_toolbox (metapackage)
-- ~~  - interbotix_ros_xslocobots (metapackage)
-- ~~  - interbotix_ros_xslocobots_examples (metapackage)
-- ~~  - interbotix_xslocobot_moveit
-- ~~  - irobot_create_description
-- ~~  - interbotix_landmark_modules
-- ~~  - interbotix_tf_tools
-- ~~  - interbotix_xslocobot_descriptions
-- ~~  - interbotix_xslocobot_control
-- ~~  - interbotix_xslocobot_joy
-- ~~  - interbotix_xslocobot_moveit_interface
-- ~~  - interbotix_xslocobot_perception
-- ~~  - interbotix_xslocobot_ros_control
-- ~~  - interbotix_xslocobot_gazebo
-- ~~  - interbotix_xslocobot_nav
-- ~~  - interbotix_xslocobot_landmark_nav
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'interbotix_common_modules'
-- ==> add_subdirectory(interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_common_modules)
-- +++ processing catkin metapackage: 'interbotix_common_toolbox'
-- ==> add_subdirectory(interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_common_toolbox)
-- +++ processing catkin metapackage: 'interbotix_ros_xslocobots'
-- ==> add_subdirectory(interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_ros_xslocobots)
-- +++ processing catkin metapackage: 'interbotix_ros_xslocobots_examples'
-- ==> add_subdirectory(interbotix_ros_rovers/interbotix_ros_xslocobots/examples/interbotix_ros_xslocobots_examples)
-- +++ processing catkin package: 'interbotix_xslocobot_moveit'
-- ==> add_subdirectory(interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_moveit)
-- +++ processing catkin package: 'irobot_create_description'
-- ==> add_subdirectory(create3_sim_ros1/irobot_create_common/irobot_create_description)
-- +++ processing catkin package: 'interbotix_landmark_modules'
-- ==> add_subdirectory(interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_landmark_modules)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Installing devel-space wrapper /home/locobot-admin/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_landmark_modules/scripts/landmark_finder to /home/locobot-admin/interbotix_ws/devel/lib/interbotix_landmark_modules
-- Installing devel-space wrapper /home/locobot-admin/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_landmark_modules/scripts/tf_map_to_landmark to /home/locobot-admin/interbotix_ws/devel/lib/interbotix_landmark_modules
-- Installing devel-space wrapper /home/locobot-admin/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_landmark_modules/scripts/landmark_manager to /home/locobot-admin/interbotix_ws/devel/lib/interbotix_landmark_modules
-- +++ processing catkin package: 'interbotix_tf_tools'
-- ==> add_subdirectory(interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_tf_tools)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'interbotix_xslocobot_descriptions'
-- ==> add_subdirectory(interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_descriptions)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'interbotix_xslocobot_control'
-- ==> add_subdirectory(interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_control)
-- Using Base Type 'kobuki'.
-- Could NOT find interbotix_xs_sdk (missing: interbotix_xs_sdk_DIR)
-- Could not find the required component 'interbotix_xs_sdk'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "interbotix_xs_sdk"
  with any of the following names:

    interbotix_xs_sdkConfig.cmake
    interbotix_xs_sdk-config.cmake

  Add the installation prefix of "interbotix_xs_sdk" to CMAKE_PREFIX_PATH or
  set "interbotix_xs_sdk_DIR" to a directory containing one of the above
  files.  If "interbotix_xs_sdk" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_control/CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/locobot-admin/interbotix_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/locobot-admin/interbotix_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:320: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed`

lukeschmitt-tr, 2024-06-12:
The remote workspace should not be built in this way. It is intended to just provide description assets and launch file. We instead recommended to use the remote interbotix ws as an underlay.

psykaunot, 2024-06-13:
I followed one of your videos on YouTube and consequently removed the CATKIN_IGNORE files (two of them). Then I used the rosdep install as shown in the tutorial but it didn't work. I'll have to close the issue for now, but I'll open it in a few weeks.

closed 2024-06-13 by psykaunot