Interbotix/interbotix_ros_rovers #69: ROS 1 Quick Start Guide - Failed to load the robot in Rviz on the remote machine

psykaunot, 2024-06-12:
### Question

I was following the Quick Start Guide [https://docs.trossenrobotics.com/interbotix_xslocobots_docs/ros_interface/ros1/quickstart.html](url), on the physical robot, there is no issue but on the remote machine, when I use:
`roslaunch interbotix_xslocobot_descriptions remote_view.launch`

Rviz opened but the robot is not loaded as you can see in the picture below:
[image]

Here is what happened in the terminal:
`locobot-admin@locobotadmin:~/interbotix_ws$ roslaunch interbotix_xslocobot_descriptions remote_view.launch
... logging to /home/locobot-admin/.ros/log/1c1abc80-288c-11ef-a799-f378e58eb97d/roslaunch-locobotadmin-319373.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.30.46.92:33735/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /locobot/
    rviz (rviz/rviz)

ROS_MASTER_URI=http://10.30.46.92:11311

process[locobot/rviz-1]: started with pid [319405]
`

When I tried this command `roslaunch interbotix_xslocobot_descriptions remote_view.launch rviz_frame:=locobot/base_link` as suggested in #25 , only the base of the robot is displayed in rviz (see below) and I would like to use the robot for real-life navigation

[image]

Additionally, the troubleshooting with the IP address did not helped to save the issue [https://docs.trossenrobotics.com/interbotix_xslocobots_docs/troubleshooting.html](url) as recommended here #47 

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

psykaunot, 2024-06-12:
I forgot to mention it but:
`roslaunch interbotix_xslocobot_descriptions remote_view.launch rviz_frame:=map` and `roslaunch interbotix_xslocobot_descriptions remote_view.launch --wait` or `roslaunch interbotix_xslocobot_descriptions remote_view.launch rviz_frame:=locobot/odom` gave the same result. Which is only the white envelope like in the first picture I sent and the output in the terminal is also the same for all of them (as the one also given above)

psykaunot, 2024-06-12:
I added this to the locobot both bashrc file and now there is no issue in rviz. The robot cam even move with the 2D Nav button:

`export ROS_MASTER_URI=http://IP_OF_THE_REMOTE_PC:11311`
`export ROS_HOSTNAME=IP_OF_THE_LOCOBOT`

lukeschmitt-tr, 2024-06-12:
Everything should be good then? This issue can be closed with the fix mentioned in your latest comment?

psykaunot, 2024-06-13:
There is another issue during the navigation, related to rtabmap (error in red) and some warning about tf concerning the base_footprint. However, it doesn't prevent navigating or mapping. I'll probably open another issue in a few weeks regarding this.

closed 2024-06-13 by psykaunot