Interbotix/interbotix_ros_rovers #70: [Bug]: Robot not loading in Rviz

psykaunot, 2024-06-19:
### What happened?

I was trying to use the robot for mapping in Habitat-Sim. I have a model of the robot loaded in Habitat, thus I needed to have the model in Rviz.

I launch the "xslocobot_description.launch " file from my own custom launch file with the argument specified:

`roslaunch habitat_ros locobot.launch robot_model:=wx250s use_joint_pub_gui:=true`

However, I following error:
` roslaunch habitat_ros locobot.launch robot_model:=wx250s 
use_joint_pub_gui:=true
... logging to /home/psy/.ros/log/2b1bd27e-2e44-11ef-a915-c500e0d9d08e/roslaunch-renoir.math.unipd.it-788170.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: while processing /home/psy/socialnav_ws/src/habitat_ros/interbotix_xslocobot_descriptions/launch/xslocobot_description.launch:
Invalid <arg> tag: list index out of range. 

Arg xml is <arg name="arm_model" default="$(eval 'mobile_' + arg('robot_model').split('_')[1])"/>
The traceback for the exception was written to the log file` 

To solve I changed the argument in "xslocobot_description.launch" to 
`<arg name="arm_model"                         default="mobile_wx250s"/>`

It solved the issue but when Rviz opened, I had the following and I can't load the entire robot:

[image]

Additionally, If I changed the rvizframe to map or odom, the robot is not loaded.

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 1 Noetic

lukeschmitt-tr, 2024-06-21:
This is likely because you're not reporting joint states from habitat. All fixed transforms have been loaded in, but because the revolute and prismatic joints don't "know where they are", they're whited out and at the origin in RViz.

We do not support the use of habitat. I would advise looking at their documentation for further support.

closed 2024-06-21 by lukeschmitt-tr

psykaunot, 2024-06-21:
But normally, if I'm not using any simulation environment (Gazebo, Habitat, whatever), it should be able to be loaded in Rviz

lukeschmitt-tr, 2024-06-21:
If you'd like to just display a virtual model of the robot in RViz, you can set the use_joint_pub_gui launch arg to true. This launches the [joint state publisher GUI](http://wiki.ros.org/joint_state_publisher#GUI) that can be used to set the joint state of any non-fixed joint on the robot.

psykaunot, 2024-06-21:
You're right I completely forgot about that parameter. However, why is the camera not loaded if the arm isn't?
Additionally, is it possible to remove the arm and use it? I tried several times to remove the arm arguments from the description file and the locobot.urdf.xacro but without success

lukeschmitt-tr, 2024-06-21:
The camera is mounted on a non-fixed pan-tilt mechanism.

To remove the arm, you can set the `robot_model` launch arg to `locobot_base`.

psykaunot, 2024-06-21:
I tried a few time but always ended up with this error:
`
 roslaunch habitat_ros locobot.launch robot_model:=locobot_base
... logging to /home/psy/.ros/log/a40bfec8-2fea-11ef-a915-c500e0d9d08e/roslaunch-renoir.math.unipd.it-881482.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: while processing /home/psy/socialnav_ws/src/habitat_ros/interbotix_xslocobot_descriptions/launch/xslocobot_description.launch:
Invalid <arg> tag: list index out of range. 

Arg xml is <arg name="arm_model" default="$(eval 'mobile_' + arg('robot_model').split('_')[1])"/>
The traceback for the exception was written to the log file
`

lukeschmitt-tr, 2024-06-21:
Please provide your launch file and any modifications you've made to the default interbotix launch files.

psykaunot, 2024-06-21:
Expect moving the file to a specific folder and thus changing the path, I think that nothing else was changed:

`
<launch>

  <arg name="robot_model"                       default=""/>
  <arg name="robot_name"                        default="locobot"/>
  <arg name="arm_model"                         default="$(eval 'mobile_' + arg('robot_model').split('_')[1])"/>
  <arg name="base_type"                         default="$(optenv INTERBOTIX_XSLOCOBOT_BASE_TYPE create3)"/> 
  <arg name="show_lidar"                        default="false"/>
  <arg name="show_gripper_bar"                  default="true"/>
  <arg name="show_gripper_fingers"              default="true"/>
  <arg name="external_urdf_loc"                 default=""/>
  <arg name="use_joint_pub"                     default="false"/>
  <arg name="use_joint_pub_gui"                 default="false"/>
  <arg name="rate"                              default="10"/>
  <arg name="source_list"                       default="[]"/>
  <arg name="use_rviz"                          default="true"/>
  <arg name="rviz_frame"                        default="$(arg robot_name)/base_footprint"/>
  <arg name="rvizconfig"                        default="$(find habitat_ros)/interbotix_xslocobot_descriptions/rviz/xslocobot_description.rviz" />
  <arg name="model"                             default="$(find habitat_ros)/interbotix_xslocobot_descriptions/urdf/locobot.urdf.xacro
                                                         arm_model:=$(arg arm_model)
                                                         base_model:=$(arg base_type)
                                                         robot_model:=$(arg robot_model)
                                                         robot_name:=$(arg robot_name)
                                                         show_lidar:=$(arg show_lidar)
                                                         show_gripper_bar:=$(arg show_gripper_bar)
                                                         show_gripper_fingers:=$(arg show_gripper_fingers)
                                                         external_urdf_loc:=$(arg external_urdf_loc)"/>

  <param name="$(arg robot_name)/robot_description" command="xacro $(arg model)"/>

  <node if="$(arg use_joint_pub)"
    name="joint_state_publisher"
    pkg="joint_state_publisher"
    type="joint_state_publisher"
    ns="$(arg robot_name)">
    <param name="rate" value="$(arg rate)"/>
    <rosparam param="source_list" subst_value="True">$(arg source_list)</rosparam>
  </node>

  <node if="$(arg use_joint_pub_gui)"
    name="joint_state_publisher_gui"
    pkg="joint_state_publisher_gui"
    type="joint_state_publisher_gui"
    ns="$(arg robot_name)">
  </node>

  <node
    name="robot_state_publisher"
    pkg="robot_state_publisher"
    type="robot_state_publisher"
    ns="$(arg robot_name)">
  </node>

  <node if="$(arg use_rviz)"
    name="rviz"
    pkg="rviz"
    type="rviz"
    args="-f $(arg rviz_frame) -d $(arg rvizconfig)"
    ns="$(arg robot_name)">
    <remap from="/clicked_point" to="clicked_point"/>
    <remap from="/initialpose" to="initialpose"/>
    <remap from="/move_base_simple/goal" to="move_base_simple/goal"/>
  </node>

</launch>

`

lukeschmitt-tr, 2024-06-21:
Running the following command with the default xslocobot_description.launch file, you should get the picture below:

```
roslaunch interbotix_xslocobot_descriptions xslocobot_description.launch robot_model:=locobot_base use_joint_pub_gui:=true
```

[image]

psykaunot, 2024-06-24:
When I use the argument with the relaunch command, it doesn't work. However, it worked when I specified the robot_model and set the use_joint_pub_gui to true. I believe that calling the description file into another launch may affect the inputs in a certain way. Thank you for your time and patience