Interbotix/interbotix_ros_rovers #71: [Question]: Locobot under other file format

psykaunot, 2024-06-19:
### Question

Hello, do you have the locobot under json format coupled with config files just like this below?
[image]

[image]

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

lukeschmitt-tr, 2024-06-21:
Could you provide a bit more context for your request? What is the json format called?

We only provide URDF-based descriptions of our robots.

psykaunot, 2024-06-21:
The json file looks like this, it is used to for defining the properties of a robot model, possibly in a simulation or robotics application:

`
{
    "render_asset": "oreo.glb",
    "collision_asset": "oreo_collision.glb",
    "join_collision_meshes": false,
    "mass": 35.0,
    "inertia": [4.704, 1.008, 4.704],
    "friction_coefficient": 0.0,
    "restitution_coefficient": 0.0,
    "COM": [0, 0.0, 0]
}
`

After you have some configurations file for definitions specific properties of the robot or its sensors:

# specify the namespace of the robot to be loaded, must be the same as the following robot_name 
robot_name: "oreo"

oreo: # robot_name

  # python class to initiate the node instance, there are: "Robot", "DepthCamera", "RGBCamera", "SemanticCamera", "LiDAR", "Laser"
  # you can also define your own sensor class in habitat_ros/script/habitat_ros/sensors/...
  # and import it in habitat_ros/script/habitat_ros/sensors/__init__.py
  type: "Robot" 

  name: "oreo"  # name of the instance, must be unique
  frame: "/base_link" # tf frame of this node

  # initial pose in the world frame
  translation: [0.0, 0.0, 0.5]
  rotation: [0.0, 0.0, 0.0, 1.0]

  # dynamic properties
  dynamic:
    mode: "dynamic" # "dynamic": physical movement, "legacy": velocity control

    navmesh: true  # moving on the navmesh only
    navmesh_offset: -0.05 # z-offset
    lock_rotation: true # lock the rotation in x and y axis to prevent falling, recommend true

    friction_coefficient: 0.05
    angular_damping: 0.3
    linear_damping: 0.3 # this could prevent the simulation from exploding

  # robot dimension properties for navmesh generation and collision, will be overwritten if collidable = True
  geometric:
    height: 1.0
    radius: 0.24

  # import 3D model
  model:
    model_path: "habitat_ros/robot/oreo/oreo.object_config.json" # folder path of where your robot modle is, e.g. {pkg_name}/robot/{robot_name}
    collidable: false  # recomment false if navmesh = true

  control:
    cmd_topic: "/cmd_vel" # topic to receive velocity command
    angular_pid: {"Kp":1.0, "Ki":0.1, "Kd":0.05}  # proportional gain for torque control
    linear_x_pid: {"Kp":40.0, "Ki":10.0, "Kd":0.5} # proportional gain for linear force control
    linear_y_pid: {"Kp":40.0, "Ki":10.0, "Kd":0.5} # proportional gain for linear force control
    maximun_force: {"x":50.0, "y":50.0, "a":1.0} # max force and torque to apply on the robot [newton, newton, newton*meter]

  # switch to true to publish ground-truth tf between map and base_link
  publish_transfromation: true
  transform:
    frame: "/map_gt" # name of the parent frame of the map, default is the frame_id of the parent node ("/map" for a Robot)
    inverse: true  # if true, the tf will be published from base_link to map, this could be used for evaluation

  # odometry configuration
  odom:
    topic: "/odometry_odm"
    frame: "/odom"
    child_frame: "/base_link"

    # the covariance matrix, currently randomly setted, could be modified if needed
    vel_cov: [
      [1.0,0.03,0.0,0.0,0.0,-0.02],
      [0.03,1.0,0.0,0.0,0.0,0.06],
      [0.0,0.0,1.0,0.0,0.0,0.0],
      [0.0,0.0,0.0,1.0,0.0,0.0],
      [0.0,0.0,0.0,0.0,1.0,0.0],
      [-0.02,0.06,0.0,0.0,0.0,1.0],]
    
  # extra tf links to be published, note that their parent is this node's frame 
  tf_links:
    base_footprint:
      frame: "/base_footprint"

      # pose of the frame
      position: [0.0, 0.0, 0.0]
      orientation: [0.0, 0.0, 0.0, 1.0]`
`

lukeschmitt-tr, 2024-06-21:
As mentioned previously, we only provide URDF-based descriptions of our robots. We additionally provide tools to spawn the robots in other simulators like [Gazebo Classic](https://docs.trossenrobotics.com/interbotix_xslocobots_docs/ros1_packages/gazebo_simulation_configuration.html).

I would advise looking at the documentation for the simulator you're trying to use to see how to do the same.

closed 2024-06-21 by lukeschmitt-tr