Interbotix/interbotix_ros_rovers #82: [Question]: Gazebo simulation error

ajinkyamar, 2025-01-22:
### Question

Hi,
When I launch gazebo simulation command, following error shows up. An Empty Gazebo simulation(with trosenrobtics office setup -walls) and rviz opens up. Why is it so? below is lg of commands. 

locobot@locobot:~/interbotix_ws$ ros2 launch interbotix_xslocobot_sim xslocobot_gz_classic.launch.py robot_model:=locobot_wx250s robot_description:=gz_classic
[INFO] [launch]: All log files can be found below /home/locobot/.ros/log/2025-01-22-06-43-47-988020-locobot-22853
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [22862]
[INFO] [gzclient   -2]: process started with pid [22864]
[INFO] [spawn_entity.py-3]: process started with pid [22867]
[INFO] [robot_state_publisher-4]: process started with pid [22870]
[INFO] [rviz2-5]: process started with pid [22872]
[ERROR] [robot_state_publisher-4]: process has died [pid 22870, exit code -6, cmd '/opt/ros/galactic/lib/robot_state_publisher/robot_state_publisher --ros-args -r __ns:=/locobot --params-file /tmp/launch_params_tzb47c5k'].

### Robot Model

locobot 250s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 2 Galactic

ajinkyamar, 2025-01-24:
could you please elaborate more? Do I need to update locobot.py in toolbox? Will it work on galactic ros2 distribution ?  Locobot.py doesnt have mention of gazebo

ajinkyamar, 2025-01-24:
Hi, I did colcon build successfully and they launched command . still same issue . Below is log

locobot@locobot:~$ cd interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_sim/launch
locobot@locobot:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_sim/launch$ ros2 launch xslocobot_gz_classic.launch.py robot_model:=locobot_wx250s
[INFO] [launch]: All log files can be found below /home/locobot/.ros/log/2025-01-24-10-26-13-286967-locobot-21292
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [21326]
[INFO] [gzclient   -2]: process started with pid [21328]
[INFO] [spawn_entity.py-3]: process started with pid [21331]
[INFO] [robot_state_publisher-4]: process started with pid [21334]
[INFO] [rviz2-5]: process started with pid [21336]

It opens up empty  Rviz and gazebo with office room.  (screenshot) 

[image]

lukeschmitt-tr, 2025-01-24:
Could you run `ros2 topic list` while Gazebo is up?

ajinkyamar, 2025-01-24:
locobot@locobot:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_sim/launch$ ros2 topic list
/parameter_events
/rosout
locobot@locobot:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_sim/launch$

ajinkyamar, 2025-01-24:
Built was done successfully as well after replacing required file

locobot@locobot:~/interbotix_ws$ colcon build
Starting >>> interbotix_xs_msgs
Starting >>> interbotix_common_modules
Starting >>> irobot_create_msgs
Starting >>> dynamixel_workbench_toolbox
Starting >>> irobot_create_description
Starting >>> irobot_create_control
Starting >>> irobot_create_toolbox
Starting >>> interbotix_common_sim
Finished <<< irobot_create_description [0.65s]                               
Finished <<< interbotix_common_sim [0.64s]
Starting >>> interbotix_tf_tools
Starting >>> sllidar_ros2
Finished <<< irobot_create_control [0.71s]
Starting >>> irobot_create_common_bringup                                    
Finished <<< dynamixel_workbench_toolbox [0.85s]
Finished <<< irobot_create_toolbox [0.83s]                                   
Starting >>> interbotix_xs_driver
Starting >>> interbotix_perception_msgs
Finished <<< irobot_create_common_bringup [0.38s]                         
Starting >>> irobot_create_ignition_plugins
Finished <<< interbotix_tf_tools [0.56s]
Starting >>> interbotix_moveit_interface_msgs
Finished <<< sllidar_ros2 [0.64s]                                         
Starting >>> moveit_visual_tools
Finished <<< interbotix_xs_driver [0.51s]                                 
Finished <<< interbotix_xs_msgs [1.64s]                                   
Starting >>> interbotix_xs_sdk
Starting >>> interbotix_xs_ros_control
Finished <<< irobot_create_ignition_plugins [0.62s]                       
Finished <<< moveit_visual_tools [0.66s]                                  
Finished <<< interbotix_perception_msgs [1.15s]
Finished <<< interbotix_xs_sdk [0.46s]
Starting >>> interbotix_xs_rviz                                           
Starting >>> interbotix_ros_xseries
Finished <<< interbotix_moveit_interface_msgs [0.94s]                     
Starting >>> interbotix_moveit_interface
Finished <<< interbotix_xs_ros_control [0.63s]
Finished <<< interbotix_ros_xseries [0.25s]                               
Finished <<< interbotix_common_modules [2.50s]                            
Starting >>> interbotix_xs_modules
Starting >>> interbotix_perception_pipelines                               
Starting >>> interbotix_common_toolbox
Finished <<< interbotix_xs_rviz [0.49s]
Finished <<< interbotix_moveit_interface [0.69s]                           
Finished <<< irobot_create_msgs [2.92s]
Starting >>> irobot_create_gazebo_plugins
Starting >>> irobot_create_ignition_toolbox
Starting >>> irobot_create_nodes
Finished <<< interbotix_common_toolbox [0.61s]                        
Finished <<< interbotix_perception_pipelines [0.75s]                  
Starting >>> interbotix_perception_modules
Finished <<< irobot_create_gazebo_plugins [0.92s]                     
Starting >>> irobot_create_gazebo_bringup
Finished <<< irobot_create_ignition_toolbox [0.95s]                   
Starting >>> irobot_create_ignition_bringup
Finished <<< irobot_create_nodes [1.09s]                              
Finished <<< irobot_create_gazebo_bringup [0.26s]                     
Starting >>> irobot_create_gazebo_sim
Finished <<< irobot_create_ignition_bringup [0.29s]                   
Starting >>> irobot_create_ignition_sim
Finished <<< irobot_create_gazebo_sim [0.22s]                         
Finished <<< irobot_create_ignition_sim [0.22s]                       
Finished <<< interbotix_xs_modules [2.48s]                            
Starting >>> interbotix_xslocobot_descriptions
Starting >>> interbotix_xs_toolbox
Finished <<< interbotix_xs_toolbox [0.18s]                                    
Finished <<< interbotix_xslocobot_descriptions [0.21s]
Starting >>> interbotix_xslocobot_control
Starting >>> interbotix_xslocobot_sim
Finished <<< interbotix_xslocobot_sim [0.25s]                                 
Finished <<< interbotix_xslocobot_control [0.28s]
Starting >>> interbotix_xslocobot_ros_control
Starting >>> interbotix_xslocobot_joy
Starting >>> interbotix_xslocobot_nav
Finished <<< interbotix_perception_modules [2.21s]                            
Starting >>> interbotix_perception_toolbox
Finished <<< interbotix_xslocobot_ros_control [0.31s]
Starting >>> interbotix_xslocobot_moveit
Finished <<< interbotix_xslocobot_nav [0.31s]
Finished <<< interbotix_perception_toolbox [0.29s]
Finished <<< interbotix_xslocobot_joy [0.39s]
Finished <<< interbotix_xslocobot_moveit [0.20s]                    

Summary: 41 packages finished [6.36s]

ajinkyamar, 2025-01-27:
Hi, could you please suggest something to resolve this issue?

lukeschmitt-tr, 2025-01-27:
It looks like you're running this on the LoCoBot. Because of the RMW configuration that's required, you may need to make sure that everything is configured correctly. Please see [this troubleshooting topic](https://docs.trossenrobotics.com/interbotix_xslocobots_docs/troubleshooting.html#can-t-see-topics-published-by-locobot-on-remote-using-ros-2) and [this configuration guide](https://docs.trossenrobotics.com/interbotix_xslocobots_docs/getting_started/rmw_configuration.html).

ajinkyamar, 2025-01-28:
Hi, 

Gazebo simulatation did open along with locobot_250s on locobot. I had to use below command
ros2 launch interbotix_xslocobot_moveit xslocobot_moveit.launch.py robot_model:=locobot_wx250s use_lidar:=true hardware_type:=gz_classic

However, after I applied force on one link, gz sim got stuck and it started to give errors like below : 

locobot@locobot:~$ ros2 launch interbotix_xslocobot_moveit xslocobot_moveit.launch.py robot_model:=locobot_wx250s use_lidar:=true hardware_type:=gz_classic
[INFO] [launch]: All log files can be found below /home/locobot/.ros/log/2025-01-28-07-28-48-936864-locobot-6717
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [6731]
[INFO] [rviz2-2]: process started with pid [6733]
[INFO] [gzserver-3]: process started with pid [6735]
[INFO] [gzclient   -4]: process started with pid [6737]
[INFO] [spawn_entity.py-5]: process started with pid [6740]
[INFO] [robot_state_publisher-6]: process started with pid [6743]
[move_group-1] [WARN] [1738070931.268481076] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1738070931.269058319] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] Error:   Link 'ee_gripper_link' specified as parent for end effector 'interbotix_gripper' is not known to the URDF
[move_group-1]          at line 423 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 555 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 555 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 550 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 550 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] [INFO] [1738070931.276743422] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1738070931.276795657] [moveit_robot_model.robot_model]: Loading robot model 'locobot'...
[move_group-1] [INFO] [1738070931.276814109] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] The root link locobot/base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[move_group-1] Link locobot/bump_caster_link had 0 children
[move_group-1] Link locobot/bump_front_center had 0 children
[move_group-1] Link locobot/bump_front_left had 0 children
[move_group-1] Link locobot/bump_front_right had 0 children
[move_group-1] Link locobot/bump_left had 0 children
[move_group-1] Link locobot/bump_right had 0 children
[move_group-1] Link locobot/button_1 had 0 children
[move_group-1] Link locobot/button_2 had 0 children
[move_group-1] Link locobot/button_power had 0 children
[move_group-1] Link locobot/camera_tower_link had 2 children
[move_group-1] Link locobot/lidar_tower_link had 1 children
[move_group-1] Link locobot/laser_frame_link had 0 children
[move_group-1] Link locobot/pan_link had 1 children
[move_group-1] Link locobot/tilt_link had 1 children
[move_group-1] Link camera_locobot_link had 1 children
[move_group-1] Link locobot/camera_depth_link had 0 children
[move_group-1] Link cliff_front_left had 0 children
[move_group-1] Link cliff_front_right had 0 children
[move_group-1] Link cliff_side_left had 0 children
[move_group-1] Link cliff_side_right had 0 children
[move_group-1] Link locobot/front_caster_link had 0 children
[move_group-1] Link locobot/imu_link had 0 children
[move_group-1] Link locobot/ir_omni had 0 children
[move_group-1] Link locobot/ir_intensity_front_center_left had 0 children
[move_group-1] Link locobot/ir_intensity_front_center_right had 0 children
[move_group-1] Link locobot/ir_intensity_front_left had 0 children
[move_group-1] Link locobot/ir_intensity_front_right had 0 children
[move_group-1] Link locobot/ir_intensity_left had 0 children
[move_group-1] Link locobot/ir_intensity_right had 0 children
[move_group-1] Link locobot/ir_intensity_side_left had 0 children
[move_group-1] Link locobot/mouse had 0 children
[move_group-1] Link locobot/plate_link had 2 children
[move_group-1] Link locobot/arm_stand_link had 1 children
[move_group-1] Link locobot/arm_base_link had 1 children
[move_group-1] Link locobot/shoulder_link had 1 children
[move_group-1] Link locobot/upper_arm_link had 1 children
[move_group-1] Link locobot/upper_forearm_link had 1 children
[move_group-1] Link locobot/lower_forearm_link had 1 children
[... 146 lines cut ...]
[spawner-9]     if not wait_for_controller_manager(node, controller_manager_name,
[spawner-9]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 95, in wait_for_controller_manager
[spawner-9]     node_and_namespace = wait_for_value_or(
[... cut ...]
[ERROR] [spawner-11]: process has died [pid 6985, exit code -2, cmd '/opt/ros/galactic/lib/controller_manager/spawner -c locobot/controller_manager diffdrive_controller --ros-args -r __node:=diffdrive_controller_spawner'].
[INFO] [robot_state_publisher-6]: process has finished cleanly [pid 6743]
[ERROR] [gzclient   -4]: process has died [pid 6737, exit code -2, cmd 'gzclient   '].
[INFO] [move_group-1]: process has finished cleanly [pid 6731]
[move_group-1] 
[INFO] [rviz2-2]: process has finished cleanly [pid 6733]

why are these spawner errors happening? Why did it not spawn again? 

Thanks

ajinkyamar, 2025-01-28:
Hi,

If I have sdf file for locobot, can it be used to create gz simulation on my desktop? what kind of changes will be required besides mesh file location, media location? does it have any other dependencies from interbotix_ws?

lukeschmitt-tr, 2025-01-28:
> ```
> [ERROR] [gzserver-3]: process has died [pid 6735, exit code 255, cmd 'gzserver [...]

Looks like gzserver is crashing. Please re-run gazebo with verbose logging.

lukeschmitt-tr, 2025-01-28:
Forwarding the Gazebo display through SSH is extremely slow. I would not recommend doing this. You may choose to use the machine directly or possibly use something like a remote desktop application instead.

ajinkyamar, 2025-01-29:
Thank for replying; Running it locally is faster. 

What does this warning imply?  

planning_scene_monitor]: The complete state of the robot is not yet known.  Missing wheel_drop_left_joint, wheel_drop_right_joint
[move_group-1] [WARN] [1738160508.319281336] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing wheel_drop_left_joint, wheel_drop_right_joint
[move_group-1] [WARN] [1738160509.338426483] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing wheel_drop_left_joint, wheel_drop_right_joint
[move_group-1] [WARN] [1738160510.351274038] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing wheel_drop_left_joint, wheel_drop_right_joint

ajinkyamar, 2025-01-29:
Hi, 
I need to apply force on individual links and join. For which, I tried setting kinematic value to true in gui. When I do this, locobot becomes unstable in gazebo. It fell down multiple times. 

How can I apply force without making locobot unstable in gazebo?

lukeschmitt-tr, 2025-01-29:
Please share an example on how you are apply forces to the robot and how exactly the robot is going unstable.

lukeschmitt-tr, 2025-01-29:
> ```
> The complete state of the robot is not yet known. Missing wheel_drop_left_joint, wheel_drop_right_joint
> ```

These are not part of the kinematic chain that MoveIt interacts with and can be ignored for now.

ajinkyamar, 2025-01-29:
if I try to apply force on link without making kinematic value for individual links to true, nothing happens. So I tried setting it to true, then the robot parts broke down in the gazebo. I managed to attach screenshot just before it wend off balance. I have added log about missing joints after robot is broken. 

[image]

[ERROR] [rviz2-2]: process has died [pid 31747, exit code -11, cmd '/opt/ros/galactic/lib/rviz2/rviz2 -d /home/locobot/interbotix_ws/install/interbotix_xslocobot_moveit/share/interbotix_xslocobot_moveit/rviz/xslocobot_moveit.rviz -f locobot/base_link --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_cahifeje --params-file /tmp/launch_params_xzo0wiju --params-file /tmp/launch_params_mft6mdzb --params-file /home/locobot/interbotix_ws/install/interbotix_xslocobot_moveit/share/interbotix_xslocobot_moveit/config/kinematics.yaml --params-file /tmp/launch_params_rg285ep6 -r locobot/get_planning_scene:=/locobot/get_planning_scene -r /arm_controller/follow_joint_trajectory:=/locobot/arm_controller/follow_joint_trajectory -r /gripper_controller/follow_joint_trajectory:=/locobot/gripper_controller/follow_joint_trajectory'].
[move_group-1] [WARN] [1738169812.482421591] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing wheel_drop_left_joint, wheel_drop_right_joint
[move_group-1] [WARN] [1738169813.508674950] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing wheel_drop_left_joint, wheel_drop_right_joint
[move_group-1] Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "locobot/left_finger_link" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (0.000000 0.000000 0.000000 1.000000)
[move_group-1]          at line 298 in /tmp/binarydeb/ros-galactic-tf2-0.17.5/src/buffer_core.cpp
[move_group-1] Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "locobot/right_finger_link" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (0.000000 0.000000 0.000000 1.000000)
[move_group-1]          at line 298 in /tmp/binarydeb/ros-galactic-tf2-0.17.5/src/buffer_core.cpp
[move_group-1] [WARN] [1738169814.542167617] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing wheel_drop_left_joint, wheel_drop_right_joint

ajinkyamar, 2025-01-29:
how to activate the camera, use it in Gazebo, and visualize it in RViz while controlling the robot with MoveIt? Scene from gazebo is not getting reflected in rviz gui. I added camera in rviz from locobot_frame_sensor but it shows blank in the image. Is it a plugin issue?

ajinkyamar, 2025-01-30:
I am trying to move locobot in gazebo using following command but it is not working

`ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args --remap /cmd_vel:=/locobot/mobile_base/cmd_vel`

below is ros2 topic list output:
```

locobot@locobot:~$ ros2 topic list
/attached_collision_object
/clicked_point
/clock
/collision_object
/display_contacts
/display_cost_sources
/display_planned_path
/goal_pose
/initialpose
/ir_opcode
/locobot/arm_controller/joint_trajectory
/locobot/arm_controller/state
/locobot/camera_controller/commands
/locobot/camera_frame_sensor/camera_info
/locobot/camera_frame_sensor/depth/camera_info
/locobot/camera_frame_sensor/depth/image_raw
/locobot/camera_frame_sensor/depth/image_raw/compressed
/locobot/camera_frame_sensor/depth/image_raw/compressedDepth
/locobot/camera_frame_sensor/image_raw
/locobot/camera_frame_sensor/image_raw/compressed
/locobot/camera_frame_sensor/image_raw/compressedDepth
/locobot/camera_frame_sensor/points
/locobot/dock
/locobot/dynamic_joint_states
/locobot/gripper_controller/joint_trajectory
/locobot/gripper_controller/state
/locobot/imu
/locobot/ir_opcode
/locobot/joint_states
/locobot/mouse
/locobot/robot_description
/locobot/sim_ground_truth_pose
/monitored_planning_scene
/motion_plan_request
/parameter_events
/performance_metrics
/planning_scene
/planning_scene_world
/recognized_object_array
/rosout
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update
/scan
/tf
/tf_static
/trajectory_execution_event

```

Could you please help me in this regard?

ajinkyamar, 2025-01-30:
For some reason, locobot is missing again in gazebo. I ran with verbose set to true. It is again just Rviz and empty room in gazebo. below is log 

```
locobot@locobot:~$ ros2 launch interbotix_xslocobot_moveit xslocobot_moveit.launch.py robot_model:=locobot_wx250s use_lidar:=true hardware_type:=gz_classic verbose:=true
[INFO] [launch]: All log files can be found below /home/locobot/.ros/log/2025-01-30-09-25-56-576504-locobot-7914
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [7933]
[INFO] [rviz2-2]: process started with pid [7935]
[INFO] [gzserver-3]: process started with pid [7937]
[INFO] [gzclient   -4]: process started with pid [7939]
[INFO] [spawn_entity.py-5]: process started with pid [7942]
[INFO] [robot_state_publisher-6]: process started with pid [7945]
[move_group-1] [WARN] [1738250758.868341008] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1738250758.869021157] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] Error:   Link 'ee_gripper_link' specified as parent for end effector 'interbotix_gripper' is not known to the URDF
[move_group-1]          at line 423 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 555 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 555 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 550 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 550 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] [INFO] [1738250758.887437747] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1738250758.887489894] [moveit_robot_model.robot_model]: Loading robot model 'locobot'...
[move_group-1] [INFO] [1738250758.887501307] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] The root link locobot/base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[move_group-1] Link locobot/bump_caster_link had 0 children
[move_group-1] Link locobot/bump_front_center had 0 children
[move_group-1] Link locobot/bump_front_left had 0 children
[move_group-1] Link locobot/bump_front_right had 0 children
[move_group-1] Link locobot/bump_left had 0 children
[move_group-1] Link locobot/bump_right had 0 children
[move_group-1] Link locobot/button_1 had 0 children
[move_group-1] Link locobot/button_2 had 0 children
[move_group-1] Link locobot/button_power had 0 children
[move_group-1] Link locobot/camera_tower_link had 2 children
[move_group-1] Link locobot/lidar_tower_link had 1 children
[move_group-1] Link locobot/laser_frame_link had 0 children
[move_group-1] Link locobot/pan_link had 1 children
[move_group-1] Link locobot/tilt_link had 1 children
[move_group-1] Link camera_locobot_link had 1 children
[move_group-1] Link locobot/camera_depth_link had 0 children
[move_group-1] Link cliff_front_left had 0 children
[move_group-1] Link cliff_front_right had 0 children
[move_group-1] Link cliff_side_left had 0 children
[move_group-1] Link cliff_side_right had 0 children
[move_group-1] Link locobot/front_caster_link had 0 children
[move_group-1] Link locobot/imu_link had 0 children
[move_group-1] Link locobot/ir_omni had 0 children
[move_group-1] Link locobot/ir_intensity_front_center_left had 0 children
[move_group-1] Link locobot/ir_intensity_front_center_right had 0 children
[move_group-1] Link locobot/ir_intensity_front_left had 0 children
[move_group-1] Link locobot/ir_intensity_front_right had 0 children
[move_group-1] Link locobot/ir_intensity_left had 0 children
[move_group-1] Link locobot/ir_intensity_right had 0 children
[move_group-1] Link locobot/ir_intensity_side_left had 0 children
[move_group-1] Link locobot/mouse had 0 children
[move_group-1] Link locobot/plate_link had 2 children
[move_group-1] Link locobot/arm_stand_link had 1 children
[move_group-1] Link locobot/arm_base_link had 1 children
[move_group-1] Link locobot/shoulder_link had 1 children
[move_group-1] Link locobot/upper_arm_link had 1 children
[move_group-1] Link locobot/upper_forearm_link had 1 children
[move_group-1] Link locobot/lower_forearm_link had 1 children
[move_group-1] Link locobot/wrist_link had 1 children
[move_group-1] Link locobot/gripper_link had 1 children
[move_group-1] Link locobot/ee_arm_link had 3 children
[move_group-1] Link locobot/ar_tag_link had 0 children
[... 52 lines cut ...]
[move_group-1] *     - GetPlanningSceneService
[move_group-1] *     - KinematicsService
[move_group-1] *     - MoveAction
[move_group-1] *     - MotionPlanService
[move_group-1] *     - QueryPlannersService
[move_group-1] *     - StateValidationService
[... cut ...]
[move_group-1] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-1] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-1] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-1] Loading 'move_group/MoveGroupMoveAction'...
[move_group-1] Loading 'move_group/MoveGroupPlanService'...
[move_group-1] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-1] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-1] 
[move_group-1] You can start planning now!
[move_group-1] 

```

lukeschmitt-tr, 2025-01-31:
What topic are you using to subscribe to the images in RViz? It's possible that the default configuration is incorrect. Could you try `/locobot/camera_frame_sensor/image_raw`?

ajinkyamar, 2025-02-02:
I tried this change as well. it still shows blank when there is some view in the gazebo environement.  (in the image tab- No image is seen) 

why is the camera in rviz aligned on z axis?

lukeschmitt-tr, 2025-02-04:
Please see if changing this line: https://github.com/Interbotix/interbotix_ros_rovers/blob/e1ab21a64c06ad2762579a4348fef77f95e9b6f5/interbotix_ros_xslocobots/interbotix_xslocobot_descriptions/urdf/pan_and_tilt.urdf.xacro#L246

to 
```xml
 <frame_name>${robot_name}/camera_depth_link</frame_name> 
```

solves this issue.

ajinkyamar, 2025-02-04:
thank you so much this solved the issue.

closed 2025-02-04 by lukeschmitt-tr