Interbotix/interbotix_ros_rovers #83: [Bug]: none of the nodes are being published after launching robot_describtion.launch file

prj281, 2025-01-28:
### What happened?

I am able to launch the xslocobot_description.launch file successfully. But after launching, the nodes are not been published. I am unable to visualize the robot in rviz. A global warning appears in rviz which says "no tf data".  I have attached the screen short of rviz and the terminal below

[image]

when we give the value of joint_state_pub as true , another error occurs which shows the invalid input type.  

[image]

could you please help us in solving this issue. 

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 2 Galactic

### Relevant log output

```shell

```

lukeschmitt-tr, 2025-01-28:
Are you running this on the LoCoBot computer, or your own machine? Please provide the output of `ros2 node list` and `ros2 topic list`.

prj281, 2025-01-29:
I am trying to do simulation in gazebo on my own computer..

[image]

prj281, 2025-01-29:
GoodEvening Sir,
I hope this mail finds you well. Currently, I am working with
locobot_wx250s. I have downloaded all the packages of ros2 galactic version
successfully.  I have given all the parameters required to launch the
description.launch file. but when i give the joint_state_pub as true, it is
throwing an error suggesting that i have used an invalid type.. but if i
have used the value of joint_state_pub as false, it will launch the rviz
but i am unable to visualize the robot and it shows global warning saying
that "no tf data. /locobot/base_link does not
exist"(pic:joint_pub_false.png). none of the nodes are getting
published(pic:node_list.png). When I give the joint_pub as true, it throws
an error(pic: joint_pub_true.png).

All these might be because of my mistake.I kindly request you to help me
figure out this issue so that we can complete our work as much as possible.
Please Sir

with regards,
Alka P