Interbotix/interbotix_ros_rovers #85: [Question]: Addition of new experiments using existing packages of locobot_250s

ajinkyamar, 2025-02-06:
### Question

**Question 1.** 

I need to add new experimental code with RL. what would you suggest to make best use of  locobot packages  ? 

**Question 2**. 

I tried to run Gazebo in my local machine without connecting to locobot. it looks like it requires actual physical base of robot on. Is there any way to work around this?  I want to run locobot's gazebo in my work computer without turning on locobot. ( remote installation part has already installed locobots packages. ) 

### Robot Model

locobot_250s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 2 Galactic

lukeschmitt-tr, 2025-02-07:
> Question 1.

Could you provide the Python script you're using? You can view the [xslocobot control demos](https://github.com/Interbotix/interbotix_ros_rovers/tree/galactic/interbotix_ros_xslocobots/interbotix_xslocobot_control/demos/python_ros2_api) for more simple examples. Each demo comes with usage instructions like the [ee_cartesian_trajectory.py demo](https://github.com/Interbotix/interbotix_ros_rovers/blob/galactic/interbotix_ros_xslocobots/interbotix_xslocobot_control/demos/python_ros2_api/ee_cartesian_trajectory.py).

> Question 2.

What issues are you seeing when trying to run Gazebo without the base turned on?

ajinkyamar, 2025-02-07:
below is what happens when I tried to run combo_control.py (from pyhton demos)  interbotix_ros_xslocobots/interbotix_xslocobot_control/demos/python_ros2_api 

```
locobot@locobot:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/examples$ python3 combo_control.py
Traceback (most recent call last):
  File "combo_control.py", line 2, in <module>
    from interbotix_xs_modules.locobot import InterbotixLocobotXS
ModuleNotFoundError: No module named 'interbotix_xs_modules.locobot'

```
Is there any issue with names of modules?

lukeschmitt-tr, 2025-02-10:
In your script, you are calling `rclpy.init`. The Interbotix Python API does this for you and double context initialization is not allowed - this is why you're seeing the runtime error.

Please also note that the Gazebo sim is not compatible with the Python-ROS API as stated in the docs: https://docs.trossenrobotics.com/interbotix_xslocobots_docs/python_interface.html

lukeschmitt-tr, 2025-02-10:
Please also make sure that you are using the ROS 2 demos. These can be found on any of the ROS 2 branches. Since you are running ROS 2 Galactic, you should use [the demos on the galactic branch](https://github.com/Interbotix/interbotix_ros_rovers/tree/galactic/interbotix_ros_xslocobots/interbotix_xslocobot_control/demos/python_ros2_api).

ajinkyamar, 2025-02-12:
Question 1.

Could you provide the Python script you're using? You can view the [xslocobot control demos](https://github.com/Interbotix/interbotix_ros_rovers/tree/galactic/interbotix_ros_xslocobots/interbotix_xslocobot_control/demos/python_ros2_api) for more simple examples. Each demo comes with usage instructions like the [ee_cartesian_trajectory.py demo](https://github.com/Interbotix/interbotix_ros_rovers/blob/galactic/interbotix_ros_xslocobots/interbotix_xslocobot_control/demos/python_ros2_api/ee_cartesian_trajectory.py).  

This is log of what happens when I run python demo in control folder 
```
locobot@locobot:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_control/demos/python_ros2_api$ python3 combo_control.py[ERROR] [1739370942.756909724] [robot_manipulation]: Failed to find services under namespace ''. Is the xs_sdk running? Shutting down...
locobot@locobot:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_control/demos/python_ros2_api$ python3 pan_tilt_control.py 
[ERROR] [1739370978.300514784] [robot_manipulation]: Failed to find services under namespace ''. Is the xs_sdk running? Shutting down...

```

ajinkyamar, 2025-02-12:
https://github.com/Interbotix/interbotix_ros_rovers/issues/85#issuecomment-2648895902 

if interbotix python api is not compatible with gazebo, How can I use gazebo to conduct rl experiements on locobot? It is essential for me to conduct custom experiments in simulation. 

Could you please suggest something to work around this?

lukeschmitt-tr, 2025-02-13:
> ```
> LookupError: Could not find the resource 'interbotix_xslocobot_moveit' of type 'packages'
> ```

The remote installation script does not install the interbotix_xslocobot_moveit package. To install it, remove the COLCON_IGNORE file from the interbotix_xslocobot_moveit package, run rosdep to install necessary package, and rebuild your workspace.

lukeschmitt-tr, 2025-02-21:
Please provide the entire console output.

ajinkyamar, 2025-02-21:
```
ajinkya@ajinkya-ThinkStation-P620:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_moveit/launch$ ros2 launch interbotix_xslocobot_moveit xslocobot_moveit.launch.py robot_model:=locobot_wx250s use_lidar:=true hardware_type:=gz_classic verbose:=true
[INFO] [launch]: All log files can be found below /home/ajinkya/.ros/log/2025-02-21-16-31-51-046722-ajinkya-ThinkStation-P620-11213
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [11218]
[INFO] [rviz2-2]: process started with pid [11220]
[INFO] [gzserver-3]: process started with pid [11222]
[INFO] [gzclient   -4]: process started with pid [11224]
[INFO] [spawn_entity.py-5]: process started with pid [11226]
[INFO] [robot_state_publisher-6]: process started with pid [11229]
[ERROR] [spawn_entity.py-5]: process has died [pid 11226, exit code 1, cmd '/opt/ros/galactic/lib/gazebo_ros/spawn_entity.py -entity robot_description -topic robot_description -x 0.0 -y 0.0 -z 0.0 -Y 0.0 --ros-args -r __node:=spawn_robot -r __ns:=/locobot'].
[INFO] [spawner-7]: process started with pid [11243]
[move_group-1] [WARN] [1740155512.449868152] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1740155512.449894451] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] Error:   Link 'ee_gripper_link' specified as parent for end effector 'interbotix_gripper' is not known to the URDF
[move_group-1]          at line 423 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 555 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 555 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 550 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] Warning: Link 'locobot/forearm_link' is not known to URDF. Cannot disable collisons.
[move_group-1]          at line 550 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
[move_group-1] [INFO] [1740155512.453447395] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1740155512.453467332] [moveit_robot_model.robot_model]: Loading robot model 'locobot'...
[move_group-1] [INFO] [1740155512.453472192] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[spawner-7] Traceback (most recent call last):
[spawner-7]   File "/opt/ros/galactic/lib/controller_manager/spawner", line 17, in <module>
[spawner-7]     from spawner import main
[spawner-7]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 28, in <module>
[spawner-7]     from rclpy.duration import Duration
[spawner-7]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/duration.py", line 17, in <module>
[spawner-7]     from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
[spawner-7]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
[spawner-7]     rclpy_implementation = import_c_library('._rclpy_pybind11', package)
[spawner-7]   File "/opt/ros/galactic/lib/python3.8/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
[spawner-7]     return importlib.import_module(name, package=package)
[spawner-7]   File "/opt/intel/oneapi/intelpython/latest/lib/python3.9/importlib/__init__.py", line 127, in import_module
[spawner-7]     return _bootstrap._gcd_import(name[level:], package, level)
[spawner-7] ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
[spawner-7] The C extension '/opt/ros/galactic/lib/python3.8/site-packages/_rclpy_pybind11.cpython-39-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-without-library-present-on-the-system' for possible solutions
[ERROR] [spawner-7]: process has died [pid 11243, exit code 1, cmd '/opt/ros/galactic/lib/controller_manager/spawner -c locobot/controller_manager joint_state_broadcaster --ros-args -r __node:=joint_state_broadcaster_spawner'].
[INFO] [spawner-8]: process started with pid [11259]
[INFO] [spawner-9]: process started with pid [11261]
[INFO] [spawner-10]: process started with pid [11263]
[INFO] [spawner-11]: process started with pid [11265]
[move_group-1] The root link locobot/base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[move_group-1] Link locobot/bump_caster_link had 0 children
[move_group-1] Link locobot/bump_front_center had 0 children
[move_group-1] Link locobot/bump_front_left had 0 children
[move_group-1] Link locobot/bump_front_right had 0 children
[move_group-1] Link locobot/bump_left had 0 children
[move_group-1] Link locobot/bump_right had 0 children
[move_group-1] Link locobot/button_1 had 0 children
[move_group-1] Link locobot/button_2 had 0 children
[... cut ...]
[ERROR] [gzserver-3]: process has died [pid 11222, exit code -2, cmd 'gzserver /home/ajinkya/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world                                                                      -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[ERROR] [gzclient   -4]: process has died [pid 11224, exit code -2, cmd 'gzclient   '].
[INFO] [rviz2-2]: process has finished cleanly [pid 11220]
ajinkya@ajinkya-ThinkStation-P620:~/interbotix_ws/src/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_moveit/launch$ 

```

lukeschmitt-tr, 2025-02-21:
> ```
> [spawner-7] ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
> [spawner-7] The C extension [...]

Please visit the (corrected) [link mentioned in the error message](https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system) for more information.

Please also note that virtual/conda environments do not play nicely with ROS and it is not recommended to use them together.

ajinkyamar, 2025-02-28:
I changed the python version(3.8). Now the error message is changed. 
```

[move_group-1] Error:   Link 'ee_gripper_link' specified as parent for end effector 'interbotix_gripper' is not known to the URDF
[move_group-1]          at line 423 in /tmp/binarydeb/ros-galactic-srdfdom-2.0.3/src/model.cpp
```