Interbotix/interbotix_ros_rovers #86: [Bug]:

anramz29, 2025-03-22:
### What happened?

I'm experiencing an error where the Python SDK cannot access some Create3 messages. I think it's a issue with the ros bridge. More concretly  unable to use the SDK as demonstrated in the pick_and_place script or the joystick package, or reset the robot's odometry as seen below. I tried reinstalling packages which worked temporarily, but the problem returned. I'm unsure if adding new packages and running catkin_make is causing this issue. My packages are here for [refrence](https://github.com/anramz29/lips_ws/tree/main/src). 

below are two examples of the issues im getting

### Robot Model

locobot_wx250s

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 1 Noetic

### Steps To Reproduce

for the first command:

source /opt/ros/galactic/setup.bash
source ~/create3_ros2_ws/install/setup.bash
ros2 service call /mobile_base/reset_pose irobot_create_msgs/srv/ResetPose {}

for the second:

roslaunch interbotix_xslocobot_joy xslocobot_joy.launch robot_model:=locobot_wx250s use_base:=true use_camera:=true

### Relevant log output

```shell
command
locobot@locobot:~$ ./reset_locobot@locobot:~$ ./reset_robot.sh 

output
ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.
ROS_DISTRO was set to 'galactic' before. Please make sure that the environment does not mix paths from different distributions.
ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.
waiting for service to become available...
requester: making request: irobot_create_msgs.srv.ResetPose_Request(pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=0.0, y=0.0, z=0.0), orientation=geometry_msgs.msg.Quaternion(x=0.0, y=0.0, z=0.0, w=1.0)))

command:

locobot@locobot:~$ roslaunch interbotix_xslocobot_joy xslocobot_joy.launch robot_model:=locobot_wx250s use_base:=true use_camera:=true

Output (note these aren't the full logs):
auto-starting new master
process[master]: started with pid [67803]
ROS_MASTER_URI=http://locobot2:11311

ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.
[INFO] [1742573418.519514028] [ros_bridge]: create bidirectional bridge for topic /mobile_base/battery_state
[INFO] [1742573418.524846423] [ros_bridge]: create bidirectional bridge for topic /mobile_base/cmd_audio
[INFO] [1742573418.528867679] [ros_bridge]: create bidirectional bridge for topic /mobile_base/cmd_lightring
[INFO] [1742573418.532323728] [ros_bridge]: create bidirectional bridge for topic /mobile_base/cmd_vel
[INFO] [1742573418.535616734] [ros_bridge]: create bidirectional bridge for topic /mobile_base/dock
[INFO] [1742573418.539212390] [ros_bridge]: create bidirectional bridge for topic /mobile_base/hazard_detection
[INFO] [1742573418.542898544] [ros_bridge]: create bidirectional bridge for topic /mobile_base/imu
[INFO] [1742573418.546578003] [ros_bridge]: create bidirectional bridge for topic /mobile_base/interface_buttons
[INFO] [1742573418.550040201] [ros_bridge]: create bidirectional bridge for topic /mobile_base/ir_intensity
[INFO] [1742573418.553450308] [ros_bridge]: create bidirectional bridge for topic /mobile_base/ir_opcode
[INFO] [1742573418.557648308] [ros_bridge]: create bidirectional bridge for topic /mobile_base/kidnap_status
[INFO] [1742573418.562637503] [ros_bridge]: create bidirectional bridge for topic /mobile_base/mouse
[INFO] [1742573418.568852799] [ros_bridge]: create bidirectional bridge for topic /mobile_base/odom
[INFO] [1742573418.574452695] [ros_bridge]: create bidirectional bridge for topic /mobile_base/rosout
[INFO] [1742573418.579238495] [ros_bridge]: create bidirectional bridge for topic /mobile_base/slip_status
[INFO] [1742573418.583176006] [ros_bridge]: create bidirectional bridge for topic /mobile_base/stop_status
[INFO] [1742573418.586516570] [ros_bridge]: create bidirectional bridge for topic /mobile_base/tf
[... 25 lines cut ...]
    from interbotix_xs_modules.create3 import InterbotixCreate3Interface
  File "/home/locobot/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/create3.py", line 7, in <module>
    from irobot_create_msgs.msg import AudioNote, AudioNoteVector, WheelTicks
ModuleNotFoundError: No module named 'irobot_create_msgs'

[locobot/xslocobot_robot-11] process has died [pid 67863, exit code 1, cmd /home/locobot/interbotix_ws/devel/lib/interbotix_xslocobot_joy/xslocobot_robot __name:=xslocobot_robot __log:=/home/locobot/.ros/log/fa186a6c-066e-11f0-947d-6d5c17325248/locobot-xslocobot_robot-11.log].
log file: /home/locobot/.ros/log/fa186a6c-066e-11f0-947d-6d5c17325248/locobot-xslocobot_robot-11*.log
[INFO] [1742573421.238193125] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/InterfaceButtons to ROS 1 irobot_create_msgs/InterfaceButtons (showing msg only once per type)
[INFO] [1742573421.541519652] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/Mouse to ROS 1 irobot_create_msgs/Mouse (showing msg only once per type)
[INFO] [1742573421.541897851] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/WheelTicks to ROS 1 irobot_create_msgs/WheelTicks (showing msg only once per type)
[INFO] [1742573421.542172701] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/WheelVels to ROS 1 irobot_create_msgs/WheelVels (showing msg only once per type)
[INFO] [1742573421.542431617] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/WheelStatus to ROS 1 irobot_create_msgs/WheelStatus (showing msg only once per type)
[INFO] [1742573421.542719652] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/HazardDetectionVector to ROS 1 irobot_create_msgs/HazardDetectionVector (showing msg only once per type)
[INFO] [1742573421.543104298] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/IrIntensityVector to ROS 1 irobot_create_msgs/IrIntensityVector (showing msg only once per type)
[INFO] [1742573421.557208413] [ros_bridge]: Passing message from ROS 2 sensor_msgs/msg/Imu to ROS 1 sensor_msgs/Imu (showing msg only once per type)
[INFO] [1742573421.573374261] [ros_bridge]: Passing message from ROS 2 tf2_msgs/msg/TFMessage to ROS 1 tf2_msgs/TFMessage (showing msg only once per type)
[ INFO] [1742573421.573627874]: Broadcasted TF from frame 'locobot/base_link' to frame 'locobot/odom'. This will only log once.
[ INFO] [1742573421.573660764]: Broadcasted TF from frame 'locobot/base_footprint' to frame 'locobot/odom'. This will only log once.
[INFO] [1742573421.573955900] [ros_bridge]: Passing message from ROS 2 nav_msgs/msg/Odometry to ROS 1 nav_msgs/Odometry (showing msg only once per type)
[INFO] [1742573421.575885568] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/SlipStatus to ROS 1 irobot_create_msgs/SlipStatus (showing msg only once per type)
[INFO] [1742573421.651238304] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/KidnapStatus to ROS 1 irobot_create_msgs/KidnapStatus (showing msg only once per type)
[INFO] [1742573421.939689705] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/Dock to ROS 1 irobot_create_msgs/Dock (showing msg only once per type)
[INFO] [1742573421.940007614] [ros_bridge]: Passing message from ROS 2 irobot_create_msgs/msg/StopStatus to ROS 1 irobot_create_msgs/StopStatus (showing msg only once per type)
[INFO] [1742573425.524823827] [ros_bridge]: Passing message from ROS 2 sensor_msgs/msg/BatteryState to ROS 1 sensor_msgs/BatteryState (showing msg only once per type)
```

lukeschmitt-tr, 2025-03-24:
Did you remove your workspace build and devel directories? If so, please remove them again and rebuild with the following commands:

```bash
source /opt/ros/noetic/setup.bash
source ~/create3_ros1_ws/install_isolated/setup.bash
cd ~/interbotix_ws
catkin_make
```

The order in which you source the workspaces/installation spaces matters when building using catkin. See [this tutorial](https://wiki.ros.org/catkin/Tutorials/workspace_overlaying) for details.

anramz29, 2025-03-24:
It worked! Thankyou